opentx/opentx #6638: OpenTX 2.2.4 - Wav files play issue

tricolil, 2019-08-18:
Radio: FrSky Taranis X9D X9D006763B01
Loaded fw: opentx-x9d-lua-luac-it-2.2.4.bin
SD card: sdcard-taranis-x9-2.2V0021.zip

I updated my old Taranis with the above written files.
The sounds (system's and custom) are played with various scratches noises.
I tested mainly the Italian language but also English one seems to have the same issue

Reinstsalled the previous condition: 2.1.9 fw and related SD card content, the issue dissapeared.

Many thanks for your support.

3djc, 2019-08-18:
We have several thousand downloads of 2.2.4 for your specific radio model and you are the first one to report something like that, so it is a very specific issue. Maybe you could do a video with sound of what you experience, to try help diagnosing this ?

bsongis, 2019-08-18:
Perhaps the SD card which is a little bit old, and OpenTX 2.2 which uses more resources?

tricolil, 2019-08-19:
Thank you for the prompt replay.
I'll try to collect more details (video, audio, ...) and I'll be back soon.

kilrah, 2019-08-19:
Just try another card first.

raphaelcoeffic, 2019-08-24:
@tricolil any new details or should we close?

tricolil, 2019-08-24:
Sorry, I'm in vacation, that's why I couldn't update with new data.
I'll be able to supply additional informations the first week of September.

Thank you again for your attention and support on this issue.

tricolil, 2019-09-05:
Hi
   sorry for the late update.
I took some time more to make some tests.
I tried to install again the 2.2.4 fw but the problem didn't come up again
even after some trials.
Actually, I changed the SD card (the one used last time was lost) so I have
the suspect that the SD card poor performances were causing the audio issue.

Let me thank you again for your support.

Kind regards
Luca

closed 2019-09-05 by kilrah