opentx/opentx #6643: Binding

craigywsm, 2019-08-19:
This is not an issue but more of a feature request.
Would it be possible to when binding for the transmitter to show on the screen a successful bind showing the receiver model it bound to?

3djc, 2019-08-19:
receivers are bound to the radio (not models) using their rxnum. Any model using same rxnum will be bound to that receiver too (and rxnum duplicates are already shown), unless I'm misunderstanding your issue ?

craigywsm, 2019-08-19:
Hi, thanks for the quick reply 3djc. Sorry i come from a non open tx background. 
Most receivers that get bound on other platforms will show on the radio once bound.
[image]
 ie does not show on the radio the receiver bound  at the point of the bind process.
hope this make sence

3djc, 2019-08-19:
Something like that exist for ACCESS receiver, but not for ACCST. Not sure how it would be managed tho, since you can rebind a receievr to another rxnum at any time, there is no way to know when that happens

kilrah, 2019-08-19:
This is a feature of the system you're using that allows the radio to know the receiver type. There is no such thing in FrSky's D8 and D16 systems, so of course we can't show anything. 

FrSky ACCESS has something like that and sure enough you can see your receiver there.

closed 2019-08-19 by kilrah