opentx/opentx #6650: Mac Simulator Momentary switches always down on x-lite-pro.

fabioeh, 2019-08-20:
**Describe the bug**
Mac Simulator Momentary switches always down on x-lite-pro.
SE and SF report down and do not go to the UP state.
Special Functions using the SE and SF trigger stuck on the down action.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup a Special Functions to trigger a sound wave file on SE or SF down repeat every 5s..
2. Run the simulator.

**Expected behavior**
Play sound only when switch is down.

**Bug behavior**

Plays sound when switch SE and SF is in the up position.

**General information**
 - OpenTX version (2.3 or N66)
 - Hardware (Frsky X-Lite-Pro)
 - OpenTX Companion (2.3 or N66)

bsongis, 2019-08-20:
@3djc would you mind checking on linux or windows? no reason it's different on MacOS

fabioeh, 2019-08-21:
Same problem on Mac and PC

raphaelcoeffic, 2019-08-24:
The issue was actually showing up on **any** momentary switch on **any** TX simulation.

PR #6662 (2019-08-26): fix #6650
switch up is -1

closes #6650
commit 274528ee (2019-08-26): Merge pull request #6662 from opentx/fix-6650

fix #6650
commit b761f7b0 (2019-08-26): switch up is -1

closes #6650
closed 2019-08-26 by bsongis