opentx/opentx #6654: Cant read models and settings

kaindub, 2019-08-22:
Radio Taranis X9 plus
Boot loader 2.2.3
Companion version 2.2.4
Windows 10

When I try to read models and settings i get this screen
[image]

When I revert to Companion 2.2.3 I am able to read models and settings

3djc, 2019-08-22:
Not sure I understand, you put a 2.2.3 screen with 9xr pro selected and you say your issue is 2.2.4 with x9d+ ?

kaindub, 2019-08-22:
My radion is a Taranis X9d plus. It has Bootlaoder 2.2.3 loaded on it
The Open TX Companion that I have a problem with is version 2.2.4, which I was requested to download by Open Companion 
version 2.2.3 of Open companion has no issues for me

3djc, 2019-08-22:
You still haven't given any details as to what your problem is !

kaindub, 2019-08-22:
When I try to read models and settings i get this screen
[image]

The models and settings do not load to the PC and the screen above  just stays

3djc, 2019-08-22:
But that screen doesn't come from 2.2.4 ! Could you take a screenshot of what actually happens using 2.2.4 ?

kaindub, 2019-08-22:
Alright its not the screen from 2.2.4
But i dont know how to take a screen shot. But I can assure you that despite this being a screenshot from 2.2.3, This is what comes up for me except it says 2.2.4 on the title bar

raphaelcoeffic, 2019-08-22:
@kaindub as it seems, the wrong radio profile is selected. Please select the right one (X9D+) and try again.

3djc, 2019-08-22:
Given he doesn't know how to take screenshots, we can't even assume this screenshot comes from his pc

kaindub, 2019-08-22:
Ok pardon my ignorance. I have never had to select the radio  profile when I have updated Open Companion version.
How do i do that?

raphaelcoeffic, 2019-08-22:
Use the marked Icon ;-) The Tooltips are very helpful as well, use them! The one for that icon reads "Radio profile".

[image]

closed 2019-08-22 by raphaelcoeffic