opentx/opentx #6657: Nightly firmware download not working in companion

skaman82, 2019-08-23:
Running Companion 2.2.4N707, have the checkbox for nightly buids checked, but getting an error trying to download the latest version.

Error transferring https://downloads.open-tx.org/2.2/nightlies/firmware/getfw.php?fw=opentx-xlite-multimodule-noheli-lua-flexr9m-en.bin - server replied: Not Found.

kilrah, 2019-08-23:
We had decided to disable nightlies for 2.2.4 since no more 2.2 release is planned.

3djc, 2019-08-23:
You should not be trying to download 2.2.4 nightlies, as no more devs is happening in 2.2 as we are not planning any new 2.2 version, as @kilrah just mentionned, I have hovewer re-enabled the download just to prevent the error

closed 2019-08-23 by 3djc

skaman82, 2019-08-23:
@3djc thanks. I was actually just testing my new R9M Lite PRO module. But I still can't select any options for output power - just the frequency. When the R9M Lite PRO module will be supported?

kilrah, 2019-08-23:
It will never be supported in 2.2, only in 2.3.

skaman82, 2019-08-23:
@kilrah okay, good to know!