opentx/opentx #6660: Play Value sometimes gives "zero" without any unit

notlebowski, 2019-08-23:
On my DLG setups I try to keep voice alerts to a minimum and extensively use a single momentary switch for multiple situations.

When I flip the momentary for less than 0.3s it will tell the value of timer1 (flight time) and the timer continues. If it's longer than that the timer stops. If it is held more than a second and a half, the timer resets to 0 and the prelaunch is started.

Most of the time everything works as I expect it to, but sometimes I hear an occasional "zero" without any unit.
I now have changed my setup in a way that it doesn't do a call-out of a timer value whenever it is lower than 10 seconds. I have yet to to fly with this changed setup, but I expect it is solved.

But what I don't understand is how any play value is ever done without a unit. Even if a value is "zero", should it not always be followed by "seconds" or "metres" independent of my setup?

I think it happens whenever the timer is reset and the timer callout is still in the queue. 

This is happening for years and many different versions. I never had another value than "zero" without a unit.

bigtwin, 2019-09-09:
I experienced the same symptom on a total different occasion. Using telemetry with a UnisensE sensor, I also just heared a naked "zero". It seems to be OK, when the battery of my power boat was full und fresh. It started somewhere during the drive and the lower the battery was, the more frequent I heard this naked "zero".
I felt like this was related to telemetry alerts which were generated on the sensor (required for M-Link) and adjusted the sensor configuration to not generate any alerts (alerts are generated only in transmitter now).
I have not tested it since I did this change. I just wanted to add it here as it may is of help to anybody.

bigtwin, 2019-09-14:
Meanwhile I have done further testing - the naked zero is not related to the Unisens E sensor and its configured alerts. I have disabled all sensor alerts and my OpenTX 2.2.4 still comes up with a plain zero. The less battery the more often. 
In oppostie to thie issue starter I cannot say which triggers this zero. I dont know (yet) whether I have a "Play Value" special function which causes this.

bigtwin, 2019-09-17:
The occurrence of the plain and naked zero is somehow mystic …

I took a log file during the drive with my speedboat while my Taranix QX7 started to speak „Zero“ once in a while. Back to my computer, I ran a simulation using Companion 2.2.4 based on the log file (one value per second). To my surprise, the naked zero was clearly reproducible in companions telemetry simulation!  :-)

I could not see any value which changes in a suspect manner. But the naked zero occurs at certain positions. It was definitely related to an active _Logical Lwitch (L01)_. A _Special Function_ based on this L01, was supposed to **„Play Value“ VFAS**. Now I was sure that my problem is the same as the issue starter reported.

However, I could see in simulation that L01 turned active, but I could not see any relevant change of VFAS (Voltage of my power battery). I decided to take another log file in 0.1sec steps. It may would show whether VFAS changes in a significant way for short spike.

Unfortunately the plain zero did not reoccur during my next boat drive, so I could not take a 0.1sec log file :-(

I order to report here what I have, I checked again the log file I had. To my surprise the naked zero was **not reproducible this time**. While the Companion simulation showed it yesterday, it did not this time. How could this be possible? I was not aware of any changes!

The only change I might have done to the model „Speedboat“ of my Taranis, is the change of some configuration value which is not supposed to be related to L01 and the according _Special Function_.

I keep prepared to gather a 0.1sec log file the next time I hear the naked zero. If I can catch it again and found it reproducible in Companions Telemetry simulation, I will also take a copy of the model „Speedboat“ and upload it here.
If anybody has an idea of what data might be relevant or should be also gathered in order to troubleshoot this, please let me know.

bigtwin, 2020-03-15:
Its being a while since I heard this "naked zero". I dont know whether it is still there, because I havnt used my race boat for while.
Its does not seem like anybody else has something to report or is interested in this failure or even an analysis. So I dont care either anymore.

bigtwin, 2020-03-17:
Zero is back again. I had massiv occurances of the naked zero. Still no idea where it comes from. A buddy of mine, who is also using OpenTX did also not had any idea. I feel like it may is related to the sensors in use. Today I had a GPS and an UnisensE connected to a X4R in my plane. However, on another plane, I only have a UnisensE and no issue with the naked zero.
I will try again to gather a logfile which will result in the naked zero when replayed on the simulation in companion.

bigtwin, 2020-11-13:
I had again a naked zero. Meanwhile on OTX 2.3.10 and a different model. 
This time I could quickly identify the reason for the naked zero: it was a _Special Function_ which "_Played Value_" of a non existent telemetrie sensor. The sensor was lost for some reason I dont remember. It could have been a deletion and re-scan for new sensors. Maybe a sensor got lost, when the physical device is not present for a while.

I dont have any problems anymore with the model I was using when I replied to this issue more than a year ago. I cannot say whether it was the same reason by than, but I dont think so. I am convinced that I would have noticed an empty field where telemetrie sensor is expected.

However, it seems not to be a bug (at least in OTX 2.3.10) but rather a feature request. It may be helpful to have something which reminds you on such empty fields in a _Special Function_. Either in Companion or in the transmitter.

If anybody has an issue with his transmitter lady simply says "zero", I would recommend to check for mal-configured _Play Value_ Special Functions and may try a deletion and re-scan for sensors. Make sure to adjust all LS and SF, in case your sensors have different reference after that.

From my point of view this issue is resolved. I leave it up to the moderators and admins to may turn it into some feature request.

Thx for OpenTX.

3djc, 2020-11-13:
Just for precision, sensors are not forgotten even if not seen for a long time, but yes, new discovery will create this.

closed 2020-11-13 by 3djc