opentx/opentx #6663: Some of the text and line colors change to white instead of black on Horus X10/X10S

robertw2a, 2019-08-25:
**General information**
OpenTX 2.2.4
Horus X10 and X10S
Using default theme.

**Describe the bug**
Some of the text and line colors change to white instead of black on Horus X10/X10S

**To Reproduce**
Steps to reproduce the behavior:

If you have INAV Telemetry LUA (1.7.1) on the display at any time during the current power up session it will cause some of the text, lines and graphic symbols to change to white instead of black. 

1) If you go to inputs screen the all the weights except the highlighted weight text is now white on a white background. If you then edit one of the inputs the graph line is also white on a white background as well. 

2) If you change models in this state the model names will be white on a white background.

3) When the radio is powered off in this state, the power down circular symbols are now white instead of black.

If the radio is powered up and INAV Telemetry LUA has not been on the display during this power up session, all text, lines and graphic symbols remain their correct color (black).

**Expected behavior**
text, lines and graphic symbol colors remain black.

raphaelcoeffic, 2019-08-26:
The issue is that the script uses `setColor()`, which changes the system colors permanently (see https://github.com/iNavFlight/LuaTelemetry/search?q=setColor&unscoped_q=setColor).

This is actually only meant for changing the colors from within a LUA theme for the Horus, not for other scripts.

Please file an issue at https://github.com/iNavFlight/LuaTelemetry and link to this issue.

closed 2019-08-26 by raphaelcoeffic

lshems, 2019-08-28:
You can work around this by storing the current color for a type using getcolor, changing it using setcolor, draw, and change it back to the stored color. 

It seems to work for me.

raphaelcoeffic, 2019-08-28:
@lshems yes, I was thinking about it. But basically it is so much easier for the script itself to just specify the colors by each drawSomething call, isn't it? Or is there any downside? Are you actually using it? (I mean, `setColor`)

kilrah, 2019-08-28:
He's using it the way he says, saving and restoring the color in the script.

lshems, 2019-08-28:
Look, the idea of a color style sheet is nice. It allows for easy changing of themes, including custom widgets.

Only problem is that there is no style sheet documentation (which style does what) and I don't think it is followed properly even in the basic theme.

I have tried to use the style sheet elements, but to many style elements have the same color, and when I change them, the standard UI becomes useless.
In short, not practical as it is right now.

So I just added the color backup, change and restore feature to the lcd.xxx functions I created, that use scaling as well based on widget size, and replace the existing ones.

This allows to have one script to be run in different widget sizes without adaptation of the code.
After a lot of fiddling works wonderfully, even to parse telemetry scripts in a widget.

Check my blog on RCgroups if you like the idea.

raphaelcoeffic, 2019-08-28:
> Only problem is that there is no style sheet documentation (which style does what) and I don't think it is followed properly even in the basic theme.

That's true. Unfortunately, not something we can change for 2.3. But there will be a new color UI in 2.4, and I hope we can improve some stuf there. I'll have a look at your ideas for sure.