opentx/opentx #6664: Crossfire micro TX telemetry show wrong power value

RookFPV, 2019-08-25:
When I set CSRF to 250mw, telemetry shows 0mw.

Crossfire micro TX 2.93 firmware
OpenTX 2.2.4
Taranis X9D Plus SE

3djc, 2019-08-25:
Are you talking about TPWR sensor ? I get the right values here, so we are likely to ned more details

RookFPV, 2019-08-25:
When I set the crossfire 25mw or 100mw TPWR shows good, when I set the crossfire at 250mw TPWR shows 0mw.
it's a meter problem.

kilrah, 2019-08-25:
TPWR displays what's received from the module, so seems to be more of an issue with the module. Is the module firmware up to date?

RookFPV, 2019-08-25:
Yes, the current modul is the new hardware 1.03 firmware 2.93 ... many people on the TBS forum write about this problem after upgrading opentx 2.2.4

3djc, 2019-08-25:
Well I have a LITE, and can set it up to 2W and I still have proper TPWR, so it is very strange

RookFPV, 2019-08-25:
The problem concerns only MicoTX and only opentx 2.2.4. if you want to check it out on the faceebok Group "TEAM Blacksheep LOUNGE" a lot of people write about this problem but no one knows solutions.
for example here https://www.facebook.com/groups/BlackSheepLOUNGE/search/?query=TPWR%20show%200mw&epa=SEARCH_BOX

3djc, 2019-08-25:
I don't do facebook, so that wont help me. Looking at the code, there is really no obvious change between 2.2.3 and 2.2.4 that  should create that.

Is 0 displayed at close range or at distance (I'm guessing the micro is adapting power too ?)

RookFPV, 2019-08-25:
When I set the Crossfire micro TX to 250mw everything works the signal is stronger, the only problem is the TPWR indicator on telemetry which shows 0mw

3djc, 2019-08-25:
It would be interesting to know if it works with 2.3 nightlies, since we are not planning to release any further 2.2.x. Anybody in the facebook group tried that ?

RookFPV, 2019-08-25:
The update for Crossfire micro TX, which allows the use of 250mw, was released in 2018.08.03.
and the OpenTX meter has not been adapted to it, only the TPWR meter does not work and only when it is 250mw, but you can see that the signal is stronger

3djc, 2019-08-25:
You are mixing things. From your own words, 2.2.3 was working fine, so there is no need to "adapt to it". There migh be a but either in OpenTX or Crossfire, but need to change anything

RookFPV, 2019-08-25:
Yes opentx 2.2.3 showed TPWR 250mw
openTX 2.2.4 shows 0mw
only 25 and 100mw shows correctly

RookFPV, 2019-08-25:
[https://ibb.co/n8mrFmH](url)

kilrah, 2019-08-25:
Nothing has changed in the affected code between 2.2.3 and 2.2.4
It's not an issue with displaying 250 since you say other modules work, and there is no difference between modules in opentx
So it seems to be a bug in the current micro tx firmware that causes it to send 0.

We'll talk to TBS about it.

RookFPV, 2019-08-25:
It is possible that the fault is on the TBS side, but from what you see 250mw increases the power of the transmitter, the picture shows the difference 1RSS & 2RSS after increasing the power, the drone was behind two walls.

kilrah, 2019-08-25:
That doesn't mean in any way that what they send back on the telemetry necessarily matches.

RookFPV, 2019-08-25:
People on the TBS forum write that you can use 250mw without any worries, they work well, only the meter shows 0mw

RookFPV, 2019-08-25:
I assume the reason this happens is because when the Lua script was written the micro TX did not have 250mW as an option. So when the module reports it is on 250mW the Lua script (the translator between opentx and tbs) doesn't know how to show it.

3djc, 2019-08-25:
You assume a lot without the right informations. Lua is not involved here at all, and the value is displayed as sent (no translation of any kind). I have contacted TBS so we work together on this issue.

RookFPV, 2019-08-25:
I'm not a programmer, I'm just speculating, I would like to help solve this problem, but the only thing I could do was to report it here, thank you for being interested in it :-)

JaceRyczak, 2019-08-26:
I have this same issue on openTX 2.1.9 on my Taranis QX7.

lichtl, 2019-08-26:
Maybe related to this?

https://github.com/opentx/opentx/blob/c9581630ffb8fc20e445ffc080debbe37aab155a/radio/src/telemetry/crossfire.cpp#L132

closed 2019-08-26 by 3djc

RookFPV, 2019-08-26:
Where is the "Crossfire.cpp" file? I don't see it on my SD card

kilrah, 2019-08-26:
Yep it seems the power is an index rather than a value for some reason. So maybe adding a 250 at the end might work since they obiously didn't shift the values. Anyway better wait for TBS's answer so we're sure of how it's supposed to work and what other values might be needed.

3djc, 2019-08-26:
So those pretending it was working in 2.2.3 where, well, not quite telling an accurate story. TBS confirmed they have added a value at the end of the list, fixing it in tomorows 2.3 nightly

RookFPV, 2019-08-26:
Yes I said 2.2.3 was good and 2.2.4 was bad

3djc, 2019-08-26:
No, it will  be fixed in tomorow 2.3 nightly. We are not currently planning to have a new 2.2x release, so not fixed there. since the next version will be 2.3, not 2.2.5 (unless a critical issue is found)

RookFPV, 2019-08-26:
So flying at 2.2.4 and 250mw dynamic is safe ?

3djc, 2019-08-26:
It is a display thing only, but you should be monitoring link quality and strength metrics

RookFPV, 2019-08-26:
Chciałem wiedzieć kiedy zachodzą zmiany za pomocą logicznych switch.
LO1 a=x 25mw play sound beep1
LO2 a=x 100mw play sound beep2
LO3 a=x 250mw play sound beep3
Now when I set LO3 a = x 0mw it will not be safe

kilrah, 2019-08-26:
Then wait for 2.2.3 and upgrade.

RookFPV, 2019-08-26:
Probably still a long way before 2.3? but great that it will be fixed :)

commit 6d7a852c (2019-08-26): Fix #6664