opentx/opentx #6666: Taranis x9-Lite running on Nightly 2.3.0 - Need Multiprotocol for MPM Lite

CDM040971, 2019-08-27:
Hi Folks, 
I seem to be stuck. Pascal has mentioned that I need to add Multiprotocol through companion to be able to use the MPM Lite Module. 

I currently have the 2.3.0 N32 night firmware firsion running which provided ACCST D16 protocols. 

How do I enable MULTI so that I can use this MPM Lite module? 

Thank you
Chris

raphaelcoeffic, 2019-08-27:
Please note this is not a support platform.

closed 2019-08-27 by raphaelcoeffic

raphaelcoeffic, 2019-08-27:
You should find good help on rcgroups.com or openrcforums.com

raphaelcoeffic, 2019-08-27:
This is also a good place to start: http://rcdiy.ca/opentx-guide/

3djc, 2019-08-27:
Also worth noting that multimodule, crossfire and luac are now included by default in all 2.3 firmwares, and therefore do not appear in the list anymore

CDM040971, 2019-08-27:
Thank you Raphael - Apologies. I incorrectly thought it was. Please could you point me to the support platform?