opentx/opentx #6667: x7 : 2.2.4 -> 2.3 won't turn on

3djc, 2019-08-27:
Starting eeprom:

[eeprom-x7-2.2.4-CRSF.bin.zip](https://github.com/opentx/opentx/files/3544720/eeprom-x7-2.2.4-CRSF.bin.zip)

Flash 2.3N69 (tested with opentx-x7-lua-en.bin)

Turn off radio

Try to turn on : you get shutdown animation and it power off

raphaelcoeffic, 2019-08-27:
I think that when loading the settings so early to have `pwrOnSpeed` & `pwrOffSpeed`, we should prevent any conversions and use default values (0) in case a conversion would be needed. Then, later, load the settings as we would have done before the "fast boot/shutdown" change.

bsongis, 2019-08-27:
of course yes!

raphaelcoeffic, 2019-08-27:
And on top of that, `eeConvert()` tries to raise a warning screen (and probably waits for confirmation....).

commit 9da190e7 (2019-08-27): Fixes #6667
PR #6668 (2019-08-27): Do not try to convert the eeprom until radio is fully ON.
closes #6667 (hopefully)
closed 2019-08-27 by bsongis