opentx/opentx #6670: X9D 2.3N69 SensorID shows 3 digits on the radio / cosmetic

Allerhopp, 2019-08-27:
[image]

raphaelcoeffic, 2019-08-27:
@Allerhopp this is absolutely normal. The instance ID now includes the module and (in case of ACCESS) the receiver concerned. The upper 3 bits of the ID might get updated based on the received telemetry packages.

closed 2019-08-27 by raphaelcoeffic

Allerhopp, 2019-08-27:
OK, thanks. An explanation for the new feature is welcome. I have only one module active and no ACCESS radio but nevertheless 3 different numbers in the 3rd digit.

raphaelcoeffic, 2019-08-27:
It's not about digits, it's about bits (unfortunately). So basically the first 5 bits are the instance ID, as it used to be (well, -1, as it now corresponds to what is really received), then 2 bits for the receiver ID and 1 bit for the module (0 internal, 1 external).

During the conversion to 2.3 on the radio, the upper 3 bits are all set, but it does not matter, just ignore them. We might actually hide them from the display (@bsongis wouldn't that be better?).

So, using  251 as an example, this is 27 if you remove the upper 3 bits.

Allerhopp, 2019-08-27:
OK, thanks again. From user view I would prefer module and RX information in the telemetry summary screen (see hardcopy) and additionally instance and ID in the sensor edit screen.

Btw. good to see, there is no more offset in instance ID.

Allerhopp, 2019-08-27:
Sorry, forgot there is another thing confusing users. It would be more clear, if instance ID/sub ID is named sensor ID. FrSky uses sensor ID too.
And what is named ID in companion better name it data ID.