opentx/opentx #6673: OTX 2.3N69 on X9D+ SE shows jitter in RC

netzwergracer, 2019-08-27:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
Jitter on the RC side on OpenTX 2.3N69 on X9D+ SE. Every 180ms a frame of 18ms is received.
Receiver is R9MM with R9M module. Both on latest firmware. FPort Flex.

**To Reproduce**
Steps to reproduce the behavior:
1. install OpenTX 2.3N69 on radio
2. Bind R9M module and R9MM receiver

**Expected behavior**
Smooth jitter-free RC input with ~6.69ms frame length
**Screenshots**
[image]
[image]

**General information**
 - OpenTX version (2.3 N69)
 - Hardware (Frsky X9D+ SE, R9MM receiver, R9M module, Fport Flex firmware)

3djc, 2019-08-28:
Have you configured failsafe the same way ?

bsongis, 2019-08-28:
Would you please double check that the same module / same receiver doesn't have the same issue on X7?

bsongis, 2019-08-28:
I close this issue, if everything above is true, it's not related to opentx

closed 2019-08-28 by bsongis