opentx/opentx #6679: 2.3 : need lua path tools for TOOLS

3djc, 2019-08-28:
We need to be able to create a ‘launcher’ that can chdir to a subdirectory and then launch a specific lua among several

raphaelcoeffic, 2019-08-29:
I had the scripts moved to the TOOLS directory and it worked ok.

bsongis, 2019-08-29:
Yes but some scripts come with a lot of subscripts, if we exported a sort of `chdir` function it would be nice

lshems, 2019-08-30:
Al boils down to the script needing to know where to find subscripts.

I just create a Lua variable containing the root path. I would never use the chair, as for me it would complicate things.

bsongis, 2019-08-30:
How do you know the root path?

3djc, 2019-08-30:
Well, the intend is to be able to use scripts as they are (of course, with scripts rewrite, you can manage it).

bsongis, 2019-08-30:
Yes, for me a chdir with a relative dir would be perfect, right?

3djc, 2019-08-30:
yes, so we can have a XfireLaunch.lua in /SCRIPTS/TOOLS that will perform a chdir to CROSSFIRE and then launch crossfire.lua (as an example)

lshems, 2019-08-30:
Ok, relocating existing scripts is another use case. Makes sense there.

Sorry for not noticing that intend.

commit 42443c5c (2019-09-10): Fixes #6679 (#6734)

Fixes #6679
closed 2019-09-10 by bsongis