opentx/opentx #6681: Model Types for Robots would help the robotics community

KrestyD, 2019-08-29:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

KrestyD, 2019-08-29:
Working with robotics we're stuck mapping channels to ailerons and the like.  And pretending tracked vehicles are v-tails or similar.  It's be nice if there was a model mode for robots (or maybe a couple). 
Even if just the companion app wizards just providing a wizard for them but mapped the results as an airplane model in the transmitter, that'd still be helpful getting jumpstarted.

bsongis, 2019-08-29:
Would you please list the modifications you would like? And describe which steps / question you would see in the "Robot" wizard?

KrestyD, 2019-08-29:
Sure, a rough take follows, but if this is feasible it might be best to do a better job collecting requirements :)

Add “Robot” to model type.  (Perhaps ground vehicle would be better)

1st wizard page for robot would be a drive type selection page

From there you need to query the drive type for the robot, common drive types for robots are
•	Car steering
•	Tracked vehicle
•	Holonomic

Car Steering wizard:

•	Throttle input (which stick, may not match the mode 1 / 2 of the transmitter)
•	Throttle channel
•	Steering input (which stick)
•	Steering channel

For car the outputs should just be 100% of the input, no fancy mixing.
Inputs and outputs should be named “throttle” and “steering” appropriately

Tracked Vehicle wizard:

Tracked vehicles can use discrete sticks for each track or a single stick mixing the channels.
•	Single Stick or Dual Stick?

Single Stick:
•	Which stick?
•	Left track channel
•	Right track channel

Output is the forward/backward value + the left/right value and the forward/backward value – the left/right value.  Eg: 100% Ail + 100% Ele and 100%E Ele – 100% Ail  (There are other techniques, but this is a simple version).  
Inputs should be labeled appropriately (throttle/steering) and the mixes/outputs should as well (left/right)

Dual Stick
•	Left track channel
•	Right track channel

Inputs and outputs should be labeled “left” and “right”.  The output is just a 100% mix of the input.

Holonomic Wizard:
Probably needs more requirements gathering, there are multiple types.  A “simple” solution is to use a stick for left/right/forward/back and another for rotation.  They can then be added / subtracted and assigned to various channels.

Another simple holonomic solution (for the tx/rx) is to pass through the left/right/spin inputs to the vehicle’s onboard computer and let it do the math.  (It also saves a channel)

Other Options?

It’s unclear to me whether additional options are common enough to include in a wizard.  Some might be:
•	A head is common (use other stick for forward/backward/left/right tilting/rotation)
•	Accessories can be assigned to switches or unused sticks – I’m less concerned about that in a wizard.

I think one of the important parts would be the naming of the inputs & outputs as many RC controlled "robot" creators don't necessarily know the normal aviation setup, stick, or channel assignments.

kilrah, 2019-08-29:
> Working with robotics we're stuck mapping channels to ailerons and the like. And pretending tracked vehicles are v-tails or similar.

For info you can rename all sticks/functions/etc already, so when you set things up manually you can name things as you want.

lshems, 2019-08-30:
I am personally in favour of having opentx as a-functional as possible. That's the power of opentx.

What you describe here is the consequence of some leftover naming conventions, that can be overcome already, as stated by kilrah, by renaming them.

A wizard is only useful if you have very clear control mixing predefined, that doesn't change other than from input mapping perspective.

Otherwise you are much better of with a complete model, and good documentation, where you use the restore functionality to have a quick point of departure.

closed 2020-05-26 by stale[bot]