opentx/opentx #6689: Problem when flashing the internal IXJT module on Horus X12S

robustini, 2019-08-31:
With OpenTX 2.3 N73 on my Horus X12S i've to insist a few times before flashing the internal IXJT module.
Here's what happens, sorry for the tilted video.
After repeated attempts I was able to write it only after having disabled it from the settings of the model.
The X12S firmware is downloaded from "Companion 2.3 N73" with multimodule, lua and luac enabled.
Since compatible I installed the IXJT module of the X10S, but this is not the problem, it also did it with the original of the X12S.
Regards,

Marco

https://youtu.be/lDRdbHyB-GQ

raphaelcoeffic, 2019-08-31:
Seems really the reset is not done properly or something like that. When you switch the module OFF before, does it work every time?

robustini, 2019-08-31:
Not always once done, but usually the first time yes, this is too random.
Sometimes after countless attempts the flashing process is ok even with the module enabled.

robustini, 2019-08-31:
Not knowing what OpenTX is going to do on the module before flashing it is not easy to understand what happens, it just does a reset and checks if the bootloader responds?

raphaelcoeffic, 2019-08-31:
From what I can see in the code, yes, that what it looks like.

robustini, 2019-08-31:
Well, but if the bootloader responds then the problem lies elsewhere.

kilrah, 2019-08-31:
All of the devs have tested it and never had a failure, and we had no other reports, so could be something specific to that radio.
Maybe a cap is larger and the module needs to be turned off for longer for it to reset correctly, no idea...

robustini, 2019-08-31:
@kilrah, does a single reset occur, or does it make multiple attempts until the bootloader responds correctly?

kilrah, 2019-08-31:
Only one.

robustini, 2019-08-31:
What do you think about repeating the reset until the bootloader responds, for at least a couple of seconds?

kilrah, 2019-08-31:
Don't think cycling power multiple times quickly is a good idea, and if the issue is a delay that's too short it wouldn't even solve the problem. 
This has never been needed and updates have been supported for years.

closed 2019-09-01 by robustini