opentx/opentx #6699: Companion 2.3 give "Can't extract Radio/Bin" whenever I try to open a 2.2.4 nightly OTX

notlebowski, 2019-09-02:
The attached otx-files can't be opened.

[image]
[OTX.zip](https://github.com/opentx/opentx/files/3566184/OTX.zip)

closed 2019-09-02 by bsongis

bsongis, 2019-09-02:
Please check after next nightly build, I have fixed an issue, and I am now able to open an eeprom generated with companion 2.2

notlebowski, 2019-09-02:
OK, I will test that tomorrow...
If it works I will check my models for differences.
Beside 3 active models there should be no difference between what's on the radio now.

If there is, I will revert my radio to 2.2.4N705, load the setup from then and then perform a new radio conversion and check for differences.

3djc, 2019-09-02:
I have just lauched a nightly creation, should be up in 90minutes or so

commit 60884b21 (2019-09-02): Attempt to fix #6699
The last file of the issue cannot be opened, but it comes from a previous nightly build?