opentx/opentx #6702: Companion N75 can now read 2.2.4 setups from my X-Lite, but makes conversion errors

notlebowski, 2019-09-02:
Companion N75 can now read 2.2.4 setups from my X-Lite, but makes conversion errors.
These errors are similar to the conversion errors made by the radio, but not the same.

This is a follow-up of https://github.com/opentx/opentx/issues/6699
The xlite-19-07-06.otx is posted in that ticket.
Here's my current 2.3 setup
[19-08-31.zip](https://github.com/opentx/opentx/files/3566994/19-08-31.zip)

If I read xlite-19-07-06.otx with Companion 2.2.4 I have in Global functions:
```FM1:Safety !L01
FM2:Spoil L13```
this becomes
```FM1:Safety !EleTrim Down
FM2:Spoil L07```

There are more differences and they can be easily spotted when you do a compare models with a recent 2.3.0 setup of mine.
You should however avoid the BAMF, Corsair and Elf as those are models I've recently changed.
The 2.3.0 setup is read with Companion 2.3 from a 2.3.0 radio.

bsongis, 2019-09-03:
I fixed an issue with all switches shifted (due to the introduction of 2 extra switches on XLite). One non regression test added, please tell me if anything is badly converted, I will fix and update the non regression test

closed 2019-09-03 by bsongis

notlebowski, 2019-09-03:
I will test this when N77 becomes available.
If that looks good I will revert my X-Lite to 2.2.4, load a setup of July and do a radio-firmware upgrade.

With Companion 2.3N77 I will then load the July config and the one that I will read from the upgraded radio. If no differences are spotted with compare it's likely that conversion errors are of the past on an X-Lite....

A question:
I am currently not using my Horus X12S that much, but can I do similar things with my X12S right now? In other words...  Do the devs consider X12S conversion to be working properly?

3djc, 2019-09-03:
yes

notlebowski, 2019-09-03:
That's quick ;-)
OK....  Then I will experiment with my X12S today and check if my models are indeed properly converted. Do I need to wait for N77 for this or were the changes X-Lite specific (I believe it's the latter)?

3djc, 2019-09-03:
N77 includes fixes for xlite only, all others should be working fine (but we are still testing x9d2019 and x10-express)

notlebowski, 2019-09-03:
OK...
Maybe I will also test my X9D+ which doesn't get any love since I bought my X-Lite.