opentx/opentx #6705: Conversion of X12S from 2.2.4 to 2.3.0N76 goes wrong

notlebowski, 2019-09-03:
I took my 2.2.4 X12S and read the configuration with Companion 2.2.4N707
I saved it as 2019-09-03-2.2.4N704.otx

I then turned off the X12S and flashed it with 2.3.0N76 and turned it on.
The screen didn't inform me of any conversion taking place. This may be normal, but contrary to what's to be expected.

I immediately knew the setup was wrong.
Every 10 seconds I get the message "Throttle disabled"
This is supposed to be done if the throttle is engaged whilst the throttle safety is on.
All the switches have been moved and there is no valid setup anymore on the radio.

When I then read the setup with Companion 2.3.0N76 I will, of course, get a lot of differences when I do a compare.
However....
Companion doesn't properly read the setup either. The configuration on the radio is different than what Companion gets after a "read model"

Here some examples with Global Functions:

**Companion 2.2.4 reading 2.2.4 radio**
SF1 ON Volume RS on
SF2 SG- Background Music Pause
SF3 !SG↓ Background Music 0choco

**Radio 2.3.0N76 after conversion (only GF1 became invalid)**
GF1 **RSSI**  Volume Vol on
GF2 SG- BgMusic ||
GF3 !SG↓ BgMusic 0choco

Companion 2.3 reading the invalid 2.3.0N76 radio (suddenly it's correct again)
SF1 **ON** Volume RS on
SF2 SG- Background Music Pause
SF3 !SG↓ Background Music 0choco

Companion 2.3 reading 2.2.4 file (suddenly it's correct again)
SF1 ON Volume **Jsy** on
SF2 SG- Background Music Pause
SF3 !SG↓ Background Music 0choco

But please repeat upgrade procedure with a 2.2.4 firmware running the 2.2.4 settings.

notlebowski, 2019-09-04:
Today I upgraded to 2.3.N78 nightly.

The RSSI that's on GF1 is now not read at all and becomes empty.
That's a bit strange as I believe there were only changes handling the X-Lite, but I'm probably mistaken there.

I'd like to stress that the changes in Global Variables are not the only changes.
Many other changes are in the individual model.
I'm only using these as an example.

**Companion 2.3N78 reading the invalid 2.3.0N76 radio**
GF1 **__** Volume RS on
GF2 SG- Background Music Pause
GF3 !SG↓ Background Music 0choco

bsongis, 2019-09-04:
@notlebowski did you start from badly converted models? or did you restore a backup?

notlebowski, 2019-09-04:
To paraphrase:
I had a "production" X12S which I didn't touch for several months.
I loaded its configuration before doing anything and saved it into the file 2019-09-03-2.2.4N704.otx

I then upgraded it to 2.3N76 which gave me an invalid setup.

I additionally wanted to note that there is more to it than just the radio conversion.

Because I was planning to test differences with Companion I also did a compare models...
Probably there's also something with Companion <=> radio as well.

bsongis, 2019-09-04:
ok, but did you restore your otx file before trying 2.3N78 ?

notlebowski, 2019-09-04:
No, because when I read the original OTX file with companion I also see a lot of differences 

If N78 has been changed regarding the conversion I can of course revert to 2.2.4, load my 2.2.4 setup and repeat the process. 
I can do this early this evening

bsongis, 2019-09-04:
Here is what I see now when opening your .otx file

[image]

Please tell me what is wrong.

bsongis, 2019-09-04:
I just checked the model "Escalibur" from your otx file. Everything seems perfectly displayed by Companion 2.3

notlebowski, 2019-09-04:
OK.
I will repeat the upgrade process and check again with N78 this evening

notlebowski, 2019-09-04:
What you proposed...
Reading my 2.2.4 setup with Companion 2.3N78 and flashing it on a 2.3N78 radio gave me a setup that was almost identical. The only reason it's not identical is Companion adding switch warnings for SI and SJ.

I didn't think this was thorough enough, so I reverted to 2.2.4 and let the radio do another upgrade and then compare the results.
Now SI and SJ are not added as warnings.
But Timer1 is now triggered by **L06** and that should be **L12**
Timer2 gets triggered by !L62 and that should have been !FM1

Another strange anomaly is that the unit "V" is suddenly omitted on telemetry field Cels

TELE3:Cels Custom Id: 300 Inst: 1 Unit: - Positive: N Log: Y 

The added feature where you can select the source of RSSI is empty after a radio conversion.
I think that should be filled with **RSSI External**.

notlebowski, 2019-09-04:
[OTX-2019-04-09-companion.and.radio.conversion.zip](https://github.com/opentx/opentx/files/3575468/OTX-2019-04-09-companion.and.radio.conversion.zip)

Strangely enough I could only see differences in the 1st model Excalibur.
I did not check all models, but at least 4 more.
I used the "compare models" feature for this.

bsongis, 2019-09-04:
Thanks for your patience
So here is the state now:

1) Companion conversion introduces switch warnings on SI and SJ
2) Radio conversion introduces a shift of 6 in the timers trigger
3) Radio conversion looses the unit "V" ? 

I don't agree for the RSSI External comment, but I will let others comment on it before deciding anything.

Agree that N76 was a big mess, we tested it on X10 ... X12S would have needed more love!

3djc, 2019-09-04:
I don’t think it is our role to guess on rssi source. Leaving it empty is a valid choice. If you want to set a specific source, then by all means do it. Anyhow it probably won’t be necessary when converting a model since that need did not exist before ACCESS

notlebowski, 2019-09-04:
>  Anyhow it probably won’t be necessary when converting a model since that need did not exist before ACCESS
Shouldn't that be a reason to agree that "RSSI External" should be chosen when an ACCST model is converted. Or is that what you're saying?

@bsongis   yes, that summary is correct.

Do note that the conversion of the timer switches went correct on most (if not all) models beside the first model.
Only the Excalibur had suddenly different triggers for Timer1 and Timer2.

It may be related to being the selected model or the 1st model. I don't know. It's merely an observation.

3djc, 2019-09-04:
No I.m saying it should be left untouched, and that it will keep working as before

bsongis, 2019-09-05:
Issue 1. bis) SI and SH should be marked as "None" in RADIO / HARDWARE I think

closed 2019-09-05 by bsongis

notlebowski, 2019-09-06:
I repeated the upgrade procedure with nightly 2.3.0N80.
The only difference compare could detect was the absence of the unit "V" in the Cells section of the Excalibur (1st and default model).
On the Me262 which also sports a FLVSS-sensor this isn't the case.

When I run the simulator and check the telemetry field there's no unit on either.

_Can you confirm seeing this difference and are you sure it's of no consequence?_

**TELE3:Cels**
Custom
Id: 300 Inst: 1 Unit: V Positive: N Log: Y 

**TELE3:Cels**
Custom
Id: 300 Inst: 1 Unit: - Positive: N Log: Y 

[OTX.N80.zip](https://github.com/opentx/opentx/files/3582727/OTX.N80.zip)

notlebowski, 2019-09-06:
BTW....
Converted models now suddenly have no RSSI source.
Does this mean they will have no RSSI-warning until they explicitly set this before flying?

Are there going to be warnings that this is the case?
If it was "RSSI external" on all models before 2.3, what's wrong with setting that?

3djc, 2019-09-06:
No, no RSSI source means it works as before (2.2.x and before). I'm thinking maybe we should hide that settings altogether unless two valid telem source are active

notlebowski, 2019-09-06:
[image]
[image]

notlebowski, 2019-09-06:
> No, no RSSI source means it works as before (2.2.x and before). I'm thinking maybe we should hide that settings altogether unless two valid telem source are active

Intuitively one would think that "no RSSI" was selected.

notlebowski, 2019-09-06:
Hiding settings based on config is IMHO a bad practice.
People (at least I) store images of screens in their brain and hiding stuff confuses.

commit e9b2a842 (2019-09-05): Fixes #6705