opentx/opentx #6707: x9e (emergency-mode and bootloder) self-initiated reboot fails

0815Creeper, 2019-09-03:
**Describe the bug**
The transmitter does not turn back on again after watchdog reset or after pressing exit in the bootloader. First batches do not seem to be affected. 

**To Reproduce**
Press exit from bootloder menue an check if transmitter turns on again or stays off.

**Expected behavior**
Transmitter turns on again normally (After bootloader-exit) or turns on again in emergencymode (after watchdog reset)

**Observed behavior**
- Transmitter turns off and stays off after bootloader-exit
- Transmitter turns off and stays off due to watchdog-reset (presumably). After pressing the pwr-btn, the transmitter turns on in ermergency-mode (! Symbol, no sd, no splashscreen)

**General information**
 - OpenTX version 2.2.3 (bootloader 2.2.3)
 - Hardware x9e

https://www.youtube.com/watch?v=8k9ilSo-q2M

3djc, 2019-09-04:
We have finaly found an X9E that does have the issue too, we are working on it

bsongis, 2019-09-04:
yes ... mine :)

bsongis, 2019-09-04:
I fixed 2.3 and I see the same issue is there in 2.2

closed 2019-09-04 by bsongis

0815Creeper, 2019-09-04:
great, thanks!

commit 995aa870 (2019-09-04): Fixes #6707 (reboot issue on X9E)
commit e291222b (2019-09-04): Fixes #6707 (reboot issue on X9E)