opentx/opentx #6712: Horus X10 not forwarding Crossfire telemetry to the internal BT module

putimir, 2019-09-05:
With (my) Horus X10S/w TBS Crossfire Micro TX module, when enabling the internal BT module, the telemetry data is not forwarded to/transmitted from the internal BT module. The feature works fine, when using internal radio, receiving s.port telemetry data. 

Steps to reproduce the behavior:
1. Use Frsky Horus X10(S) transmitter with TBS Crossfire Micro TX module 
2. Have a working connection with a model, receiving (through XF link) and displaying telemetry data 
3. Enable internal BT module and set it to "telemetry"
4. Connect with a BLE tool of choice and listen to the data stream; there is no data transmitted, when using external XF module, but it works when using internal radio and s.port as telemetry.

I would like to request, that the Crossfire telemetry is being forwarded to the BT module, either in native format or  translate it to s.port 'format' or other, whichever is more simple and faster to accomplish.

**General information**
 - OpenTX version: "opentx-x10-2.2.4"
 - Hardware: Frsky Horus X10S
 - TBS Crossfire Micro TX, version 2.41

davwys, 2019-09-05:
To add to this:
I have successfully tested FrSky's own TX modules such as the XJT or the R9M to work in this manner with the Horus and found that they forward telemetry via the Horus' built-in BT module - so this problem is most likely exclusive to third-party modules like the Crossfire.

putimir, 2019-09-05:
Thank you for rapid collaboration, just to add: I'd be using this to feed the data to the u360gts tracker: https://github.com/raul-ortega/u360gts/tree/master/wiki

kilrah, 2019-09-05:
The telemetry output from OpenTX will be straight Crossfire, which that project doesn't support.

3djc, 2019-09-05:
I will do that change, but as serial, we will output data in Crossfire native format (as we do output FrSky telem in their native format)

davwys, 2019-09-05:
The project's author is currently working on CRSF telemetry implementation (and has already uploaded videos of it working), I'm assuming it will work soon - so that shouldn't be an issue.

putimir, 2019-09-05:
@kilrah: just as davwys said: the implementation of the XF for that project is in beta stage, so, this solution will be most certainly beneficial!

3djc, 2019-09-05:
xfire telem comes at a really fast rate, it will be interesting to see if a BT ble link can cope with it

3djc, 2019-09-05:
Will be in tomorows 2.3 nightly

closed 2019-09-05 by 3djc

putimir, 2019-09-05:
Regarding bandwidth: good point; it is already suggested (https://github.com/raul-ortega/u360gts/issues/47#issuecomment-522525250) that the internal baud rate is limited to 115200; but I'm more than happy to test all the scenarios!

putimir, 2019-09-06:
YESS! Getting data! I'm using 115200 baud on the XF module. Will try later with 400000 baud. Here is the captured data, in case someone cares to take look if it is sane and usable...
[btlog.log](https://github.com/opentx/opentx/files/3582390/btlog.log)

Thank you @3djc!

putimir, 2019-09-11:
Hello, @3djc: is this included in the todays RC1?

From release notes: "Add Crossfire telemetry mirror on bluetooth"?

Thanks

3djc, 2019-09-11:
well, I think you answered the question yourself

putimir, 2019-09-11:
:D :D

Sorry, wasn't sure if it was Horus related.... Thanks again for the intervention!

Regards!

putimir, 2019-09-11:
Hey sorry to bother you, with one more thing: I have installed now the RC1 companion over the N80 version, but now it gives me the "compilation error" when I try to download the FW, for the Horus X10...

I have the profile set-up and selected. I does not give me error, if I switch to the X7 profile and download the FW...

3djc, 2019-09-11:
What exact options are you using ?

putimir, 2019-09-11:
just noheli and lua

putimir, 2019-09-11:
[image]

3djc, 2019-09-11:
I just downloaded it ok

putimir, 2019-09-11:
What do you recommend? Make a new profile perhaps?

putimir, 2019-09-11:
Yup, that fixed it...

putimir, 2019-09-11:
Can just add this: I'm having no problem getting telemetry through BT even if I set the module baud rate @400000... :D

putimir, 2019-09-12:
One thing, I noticed, regarding telemetry display, changed in this version:

With the XF transmitter modules, you get some of the sensors from TX module in, even if not connected to a receiver, for example, TPWR and RQly. In previous OpenTX, these were getting in nicely and the telemetry display, as well as all the logical switches and functions were working as intended...

Now these sensors come in intermittently, the displays are blinking, the LS are firing multiple times....

When the TX connects with an RX, these symptoms are gone, all the sensors, both from RX and TX are stable...

https://www.youtube.com/watch?v=6f6MHlQQJQY

3djc, 2019-09-12:
please open separate issues, it is impossible to track things like that, thanks

putimir, 2019-09-12:
Ok, sorry, I posted here, because I thought perhaps it's related to the BT mirror...

PR #6713 (2019-09-05): Enable dump xfire telem using BT
This closes #6712