opentx/opentx #6714: Future update requirements

unipheas, 2019-09-05:
> ‪As a reminder, Mac software distributed outside the Mac App Store must be notarized by Apple in order to run on macOS Catalina. To make this transition easier ... , we’ve adjusted the notarization prerequisites until January 2020.  

https://developer.apple.com/news/?id=09032019a‬

bsongis, 2019-09-05:
for Raphael ;)

raphaelcoeffic, 2019-09-05:
@unipheas as far as I could read, it seems this only applies to signed code, which is not the case for Companion. I hope they keep the explicit whitelisting (right-click on the app to start, confirm the exception) as it is right now.

unipheas, 2019-09-05:
Okay ... I figured I’d make you guys aware in case there are problems in the future. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

raphaelcoeffic, 2019-09-05:
@unipheas fingers crossed! otherwise we won't be the only ones in trouble... Don'T want to shell out the 100$/year for signing the code...

unipheas, 2019-09-05:
If need be I’d be willing to donate my account for the cause, just let me know.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2019-09-05 by unipheas