opentx/opentx #6716: 2.3: X9D+ simulator on Mac -

athertop, 2019-09-06:
Spotted an issue with companion Taranis x9d+ simulator both in nightly 70 and 71 and still exists in today's nightly - all related to Mac versions. I am using v20 sd card contents, and if I go into radio menu on simulator and to the new Tools page, I see the tools listed for a few moments (think there are 4 or 5 scripts listed, inc SxR, etc) then the screen blinks and those lines are replaced by a single line in slot 1 showing 2-3 characters of corrupt text. If I page back to SD card browser screen, I see it blank instead of the SD card contents. If I close and restart the sim, it gives an error about card version - as It can't see the SDCard folder contents now. A restart of the Companion app sometimes (10% of cases for me) fixes this but most often the the issue reoccurs every time.

A video of the issue can be seen here:
https://youtu.be/PcYvX7CqHW4

**General information**
 - OpenTX version N80 Companion (not tried on Radio)
 - Hardware in sim: Frsky X9D+

3djc, 2019-09-07:
Mac related, doesn't appear on Linux or Windows

athertop, 2019-09-07:
On the same Mac, I have Win10 running as a VM inside VMWare Fusion. I installed the N80 version in Windows and a fresh copy of the SD Card, and see this following the same clicks to get into Simulator Tools page:
[image]
In this case, only the Tools page gets corrupted and I can happily access the SDcard contents if I page back.
I also have a Win10 Tablet I can try this on so will update once I manage to test that.

3djc, 2019-09-07:
Then I need to understand why I can"t reproduce, will investigate

athertop, 2019-09-07:
To reiterate the process I follow which leads to this for me:
New install of Companion with no models installed
Launch Companion
Configure Profile for X9D+ pointing at SD card contents
Make sure LUA option is selected (I know this is just for firmware download but its what I have)
Then close and relaunch companion
Click + on toolbar to open a new window (which has no models in it of course)
Click the Simulate button at the top of that window
Wait for the simulator to get to New model wizard and exit back to main screen in sim (reads Model01)
Long press MENU
Press PAGE to reach Tools page
Wait 5 seconds
=screen corruption

athertop, 2019-09-15:
I have tried the latest release candidate, and this issue seems to be resolved, so I will close this ticket.

closed 2019-09-15 by athertop