opentx/opentx #6717: [2.3N81] Telemetry not working (ISRM ACCST D16)

uragano47812, 2019-09-07:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
Telemetry is not working (using ACCST D16 with last 1.1.3 firmware module). The radio doesn't see the sensors and can't find any new sensors with the discover function. Only RSSI is reported and even not the RXbat showing. I tried with different sensor like Frsky 150A sensor and Unisens-E sensor. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Telemetry setup'
2. Click on 'discover new sensors'
3. See error: no new sensors will appear and RSSI is showing "default"

**Expected behavior**
See the sensors list with their usual ID.

**General information**
 - OpenTX version 2.3N81
 - Hardware  Frsky X9D+ SE 2019 with ISRM 1.1.3 module firmware
 - OpenTX Companion 2.3N81

closed 2019-09-07 by bsongis

Talha909, 2019-09-07:
same issue faced by me.

3djc, 2019-09-08:
This should now be working in N82

commit d681ba42 (2019-09-07): Fixes #6717