opentx/opentx #6718: X9D+ conversion errors in 2.3.0N81

notlebowski, 2019-09-07:
After the X-Lite and X12S I tested my third and last Frsky radio, the X9D+
I have no more radio's to test and it turns out that all different models should be tested with real-life setups to expose any changes. On version N81 the X-Lite converts correct and although the ticket of the X12S is already closed I still see a difference that worries me. Not for the issue itself (the loss of the unit "V" in telemetry), but the fact that there's a difference at all.

I found that the X9D+ loses its curves or in INPUTS
So models having Expo on 2.2.4 will have none in 2.3.0N81

The 2 bar attributes for "Volts source" and "Altitude source" also went missing on 2.3.0N81

**2.2.4**
Top Bar
Volts source: RxBt; Altitude source: Alt
**2.3.0N81

**
Top Bar
Volts source: None; Altitude source: None

[OTX-X9D+.2.3.0N81-conversion.zip](https://github.com/opentx/opentx/files/3586405/OTX-X9D%2B.2.3.0N81-conversion.zip)

MRC3742, 2019-09-09:
I can confirm the above problems on EEPROM conversion in radio X9D+ after update from v2.2.4 to v2.30N82.

"I found that the X9D+ loses its curves in INPUTS" (on Throttle Input)
( Although my Custom curves are still retained. )
"So models having Expo on 2.2.4 will have none in 2.3.0N81"
"The other difference detected by "Compare models" are the 2 bar attributes for "Volts source" and "Altitude source". These also went missing on 2.3.0N81"  (and also in N82)

**Also on INPUT Channels that originally had expo now have no expo listed and instead have 15% or 16% offsets added? Even if the particular INPUT line had no expo initially.**

Print Model .pdf comparison files attached.
[X-Racer_2.2.4.pdf](https://github.com/opentx/opentx/files/3588187/X-Racer_2.2.4.pdf)
[X-Racer_2.30N82.pdf](https://github.com/opentx/opentx/files/3588188/X-Racer_2.30N82.pdf)

Telemetry sensor ID's are correct but Instance are decreased by a value of 1 ( 25 converts to 24 and 28 converts to 27 ) Deleting all sensors and then Discover Sensors will cure the problem with new Instance of 248 for the # 25 sensors and 251 for the # 28 sensors. 
This now works in Radio but saving to Companion will not save the new values and changes them back to 24 and 27. 
Even changing the Instance values in Companion to 248 and 251 will not transfer back to the Radio but instead writes them as 24 and 27.

Telemetry Model .pdf comparison files attached
[SkyDrone_2.2.4.pdf](https://github.com/opentx/opentx/files/3588309/SkyDrone_2.2.4.pdf)
[SkyDrone_2.30N82.pdf](https://github.com/opentx/opentx/files/3588311/SkyDrone_2.30N82.pdf)

notlebowski, 2019-09-09:
Strangely enough mine didn't produce offset changes and also cleared the custom curves.

I quickly checked the radio itself before I went to work to see if "compare models" missed it, but no. It wasn't on the radio either.

MRC3742, 2019-09-09:
Also it seems to improperly convert only the INPUTS.

In the Telemetry Model .pdf's (attached in above post) I had the throttle curve set up in the MIXERS and that converted correctly.

closed 2019-09-09 by bsongis

notlebowski, 2019-09-09:
@MRC3742 
Can you repeat the conversion process on your X9D+ to see if it now converts correctly?
It will take a few days before I'm close to mine.
I will do the conversion anyhow upcoming Friday.

MRC3742, 2019-09-09:
I downloaded the updates and built a new 2.3 firmware.bin

Updating from v2.2.4, now expo and throttle curve are converted properly.
Top bar "Volts source" converted properly (I have no "Altitude source" on reviewed models)
Also the telemetry instance is updated to the newer numbers (248 and 251) works as expected.
Radio problems seemed to be fixed from this commit but I will review more complicated setups later today.

Companion still reads telemetry setup from radio with wrong instance on each sensor but this may need another Issue to be opened?

@bsongis Thank You for the quick update and resolution for this problem.

bsongis, 2019-09-09:
No, the instance in telemetry has been changed on purpose

MRC3742, 2019-09-10:
Yes, I understand the Instance numbers have changed ( to 248 and 251 in my model? )
But the problem is when reading Radio EEPROM into Companion they get changed to 24 and 27
Then Writing them back to Radio also writes them as 24 and 27 causing telemetry to be unavailable.

This problem does not exist if EEPROM is saved in Radio using SD Card and then restoring it from the bootloader screen.  If this option is used then all works as expected.

I have also compared complex models with multiple curves, logical switches and special functions.
In the Radio these models also all appear to have converted properly from v2.2.4 to v2.3.0
Again the only issue is reading the EEPROM into Companion changes the telemetry Instance value.

commit dd25e822 (2019-09-09): Fixes #6718