opentx/opentx #6720: Strange handling of FlsSky telemetry data

Steini63, 2019-09-07:
Hello from a big openTX fan!

I've constructed a diy converter for outputting blheli_32 telemetry data to FrSky (D8 or SPort) or FlySky (iBus) and display that in an openTx Transmitter. For details  see https://www.franksteinberg.de/BLHeliTelemetryFeeder.html (german only - sorry).
Everything works in principle, but I found strange calculations of some metrics in the FlySky iBus telemetry (unlike for FrSky) :

1. Unit RPM:
1.1 "Multiplier" in the openTX sensor menu adds an offset instead to multiply.
1.2 "Blades" in the openTX Sensor menu require extremely high values to output realistic values (366 blades to convert from eRPM to RPM on a shaft with a 14-pole motor, which should be 7).

2. Unit Current (A):
2.1 openTx gets an integer 16bit value - resolution 0.1 A (for example 123 for 12.3 A). In the openTX Sensor menu "Ratio" the value 25.5 is required to divide by 10; I would expect 10 to be the required value.

The described behavior affects both openTx 2.2 and 2.3

kilrah, 2019-09-07:
There is little support for FlySky telemetry, the only sensors that are handled are here:

https://github.com/opentx/opentx/blob/2.3/radio/src/telemetry/flysky_ibus.cpp#L52

Steini63, 2019-09-08:
Thanks for the link, I looked at the source code (as far as my limited knowledge of C has made possible).

First of all for clarification: Values ​​are displayed for all sensors of my converter. Only the subsequent calculation of RAW values ​​by openTX (multiplier) is strange.

1. RPM calculation for FlySky iBus:

Here I would like to express a supposition:
In the source text of "flysky_ibus.cpp" the RPM sensor is configured as follows:
```
// RPM
  {0x02,   ZSTR_RPM,   UNIT_RAW,   0},
```
The calculation of the propeller blades and the multiplier in "telemetry_sensors.cpp" is not used because of the configuration as UNIT_RAW. Would changing UNIT_RAW in UNIT_RPMS help?
Even better, I find it, if instead of the sensor-ID the user-input of RPM as a unit would trigger the RPM calculation. Then each sensor-ID could be used for the RPM display.

2. Processing of the multiplier with other sensors (FlySky _and_ FrSky):

When entering a multiplier on openTX, the displayed values ​​are calculated according to the formula:
displayed value = RAW value * (multiplier / 255)
I would expect:
displayed value = RAW value * multiplier

I honestly do not know if this is a bug or a feature !?

kilrah, 2019-09-08:
> Would changing UNIT_RAW in UNIT_RPMS help?

Yes

> Even better, I find it, if instead of the sensor-ID the user-input of RPM as a unit would trigger the RPM calculation. Then each sensor-ID could be used for the RPM display.

It's the case already, you should be able to change the sensor unit to RPM and the ratio/offset should become multiplier/blades.

> displayed value = RAW value * (multiplier / 255)
I would expect:
displayed value = RAW value * multiplier

That's a relic based on the way FrSky's analog values work, they're raw 8-bit values that need to be translated to a user unit. User choses "what 255 means" hence the factor. 

We'd like to improve it and make it a normal multiplication factor for things other than those at some point, but it's been lingering for several years becasue nobody got a good idea of how to separate it. 

https://github.com/opentx/opentx/issues/2615

Steini63, 2019-09-09:
> It's the case already, you should be able to change the sensor unit to RPM and the ratio/offset should become multiplier/blades.

openTX shows the screen depending on user-input but does the corresponding calculation depending on the sensor-ID. So I suggest to do **both** depending on user-input 

or

Change UNIT_RAW to  UNIT_RPMS for FlySky sensor-ID 0x02

Is there any hope to get that into release 2.3 ?

closed 2020-06-05 by stale[bot]