opentx/opentx #6722: X10s Express - Cannot flash internal module with latest update

Quadpilotman, 2019-09-07:
When trying to flash the internal module with the latest update from Frsky an error is reported and the module is not flashed.

Steps

Download latest module firmware from the frsky site  ISRM V1.1.3
Enter radio setup
Select sd-hc card
Scroll to FIRMWARE folder and open.
Select ISRM_S_X10_190805.frk
Select Flash Internal Module

Receive Firmware update error  - Device not responding

General information

OpenTX version 2.3.0N81
Hardware FeSky X10S Express

br34k0r, 2019-09-08:
Same issue here...

Also I'm not able to bind any receiver

Quadpilotman, 2019-09-08:
I have bound in D16 mode no problem but have not tried access yet

br34k0r, 2019-09-08:
I see you also have the X10S Express. Did you use "FrSky Horus 10/10S" or "FrSky Horus 10 Express" as Transceiver?

Quadpilotman, 2019-09-08:
I used the Horus 10 express option as this matches my transmitter and will support the access protocol.

bsongis, 2019-09-08:
This file should be .frsk, not .frk

Quadpilotman, 2019-09-08:
oh ok I will try it.  Note  the  .frk extension is present in the download zip from the x10 express downloads page

Quadpilotman, 2019-09-08:
Ok I have changed the extension to .frsk now when I select it it just offers Copy Rename Delete as options

kilrah, 2019-09-08:
Renaming will not work, it's a different file type altogether. FrSky just posted the wrong file.

Quadpilotman, 2019-09-08:
Well you are right about that , however I assume that file would work using FrOs or is that incorrect?

bsongis, 2019-09-09:
FrSky just updated the firmware, please confirm it is OK now, thanks!
https://www.frsky-rc.com/horus-x10s-express/

closed 2019-09-09 by bsongis

Quadpilotman, 2019-09-09:
Hi
Just tried this it still does not work seems to me it's the same file maybe i will do a file compare just to see.

Quadpilotman, 2019-09-09:
ok just done an fc /b on the file I just downloaded compared to the one I downloaded 4 days ago and there are no reported differences so either Fsky have not changed the file or they have duplicated the error and put the same file on, more likely though is an error in the flashing code as it reports "Device not responding" as the firmware update error please reopen the fault

bsongis, 2019-09-09:
did you extract the FRSK file (not FRK)
I just checked it, the file seems good

Quadpilotman, 2019-09-09:
I do not see an frsk file in the expanded zip that I downloaded from the link above am I getting sent to a different frsky server I am in the UK I get ISRM_S_X10_190805.frk and a readme.txt file. Even then opentx will not give an option to install firmware from an .frsk extension only file operations

Quadpilotman, 2019-09-09:
Just to confirm I am talking about OpenTx behavior here not FrOS

kilrah, 2019-09-09:
The file really has changed. 
https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/X10X10SExpress/FW-X10SEXPRESS-ISRM-V1.1.3.zip

Quadpilotman, 2019-09-09:
so what do you get when you unzip I tried again but it still won't flash to my X10 express internal module same error

Quadpilotman, 2019-09-09:
can you attach the .frk or .frsk file that actually works for you

kilrah, 2019-09-09:
Can't attach on here. This zip has an frsk. Delete any you might have downloaded previously before grabbing it, probably just a file management mistake.

Quadpilotman, 2019-09-09:
Ok so here is the issue I did what you suggested but this made no difference still getting the same file ie the same .zip is being served, then I moved to the edge browser instead of chrome, then I got the right zip. So what 's going on here is that because the file name has not changed on the web server the old file is still being held in some cache somewhere possibly the browser possibly a cache between me and the frsky server, now for a https site this is unusual behavior, the usual behavior is for the expiry expiry date of any file to be ignored by caches and indeed the browser itself but this normal web behavior is not happening so anybody using chrome (if that is the culprit) will experience this issue my guess is that Chrome is respecting the expiry date of the file set on the web server mayby Frsky could be encouraged to make all these 0 (ie immediate expiry) on files that are available for download to avoid this problem in the future it is a bit onerous to clear the browser cache for every interaction with the frsky website. I don't think the problem is external caches because a new browser instance  (edge in my case) got the correct updated file. In conclusion the new file flashed fine thank you to all concerned

bsongis, 2019-09-09:
http / https has nothing to do with expiration model, it seems normal to me that your browser considers the file cached as fresh.

next time I will ask FrSky to change the filename of the zip (as per google preconisations), it will avoid this kind of issue

Quadpilotman, 2019-09-09:
Yep change of filename will crack it, http /https does have an effect on web proxies caching behaviour , but you are right browsers have been caching https content for a while now, however pre 2010 they did not do this very much hence https created expiry by default, Setting the expiry can also solve this kind of problem browsers do respect this in my experience, but it is a bit tedious to manage however all the download areas of the frsky web site could be configured this way then filename changes become unnecessary to ensure fresh content and provides an additional safeguard in case someone forgets to change the file name.

Thanks for getting this resolved