opentx/opentx #6723: Language input error

Arizona5318, 2019-09-08:
Every time I open OpenTX Companion it open in a foreign language (not english) which I can not read.  
When I try to switch it back to english it remains in the foreign language.
I have removed OpenTX from my MacBook Pro several times, however when I reload OpenTX (ver. 2.2) it reloads in a foreign language and not english.
I also reloaded an older version from a previous back up hard drive and it loaded the older version, however it was in a foreign language and not english that was being use when the back up was created.
I contacted Apple tech support however they were unable to correct the issue on OpenTX opening up in a foreign language.
See Attached example of the foreign language.

[image]

kilrah, 2019-09-08:
[image]

Arizona5318, 2019-09-08:
I am using an Apple MacBook Pro note book computer running IOS operating system.
Is the above screen shot from a PC Windows computer or from an Apple computer?
I do not recognize the screen shot being from an Apple computer.

3djc, 2019-09-08:
It is exactly the same screen on Apple computers, this is inside Companion

Arizona5318, 2019-09-08:
My problem is all the menu tabs and menu drop downs are in a foreign language that I cannot read I require assistance getting to the correct tab and item on the drop down menu.  Can you guide me to the screen shot provided.

3djc, 2019-09-08:
Even in foreign language, the menu are is the same order/place

Arizona5318, 2019-09-08:
I have attached a screen shot of the first page of Companion when it opens, can you guide me to the first tab to select.
[image]

kilrah, 2019-09-08:
It's the same on all systems. Browse the items menu bar until you find one that looks like it and has the language list in a submenu.

Arizona5318, 2019-09-08:
I have attached screen shot of the first level of drop downs from the five top line tabs.
I have gone thru all of the drop down menu items, any suggestions which drop down menu item to try?
[image]
[image]
[image]
[image]
[image]

kilrah, 2019-09-08:
You're using 2.2.0, start by upgrading to 2.2.4.

Arizona5318, 2019-09-08:
On Sept. 8, 2019 Robert M...  wrote...

Hello Andre B...

Thank you for providing me with assistance to sort out this foreign language issue on my Companion.  I have been using Companion for two years, it has run flawlessly until about one + months ago.  That is when strange error messages would pop up upon opening Companion.  They did not appear to be any thing serious so I did not record them.  I could always close the error message and it would not or very rarely reappear again.

Had one strange one on by Taranis TX when I fired it up at the field.  It indicated that I had a faulty antenna but never indicated if it was the TX or RX antenna.  That error message concerned me so I stopped flying immediately.  Change models on the TX and fired it up on two other models I had with me at the field, no antenna error messages appeared with them, it also never appeared again on the model that gave me the antenna error message the first time.  I also range check all models including the model that had the antenna error message pop up.  That model had a lower RSSI than the other two so I stopped flying it until I have time to change out the RX.  I have had flight with the TX since, the antenna  error message never reappeared.

This language issue I have no idea where it come from.  I use an Apple MacBook Pro which has never given me problems.
Last night I spent two plus hours with Apple Care Tech Support, via phone and went thru the entire MacBook Pro running diagnostic checks, went thru setting, run it in Safe Mode, flushed out all the ram and buffers, and many other things to the point that the tech felt the machine was in good health with no malware or virus issue.

We also reinstalled Companion from the Open-Tx website with the same language issue reappearing immediately.  Reinstalled Companion from from a back up hard drive.  The reinstall from the hard drive was a version save several months back when the language issue was not present.   Did several reinstall  from must  earlier back ups dates when Companion was running perfect, again when we fired up Companion the language issue appears immediately.

Any assistance you can provide will be greatly appreciated.

Cheers

Robert M.

Arizona5318, 2019-09-08:
I have up graded to 2.2.4.  Here is a screen shot of the opening screen.
Any suggestion were to try to find the menu relating to language?
[image]

kilrah, 2019-09-08:
Looks like the Czech translation is broken and not all menus are present. 
Close companion, in the Finder browse to ~/Library/Preferences and delete the file "com.opentx.Companion 2.2.plist".

Relaunch companion and all settings will be reset to defaults. Don't select Czech again in the menu for now.

Arizona5318, 2019-09-08:
On Sept, 8, 2019 Robert M.. wrote...

Hello Andre B...

I have search for the "com.opentx.Companion2.2.plist"  as well as "com.opentx.Companion2.2.4.plist" and have not found a copy of either file.  I also just searched for "plist" found many "plist" files however did not come up with the file you mentioned.

Any suggestion.

Cheers

Robert M..

&&&&&&%%%%%%$$$$$$$$$$$#

kilrah, 2019-09-08:
Did you correctly go to `~/Library/Preferences` and not `/Library/Preferences`?

If yes try following what's mentioned there:
https://forum.qt.io/post/203260

Log out, log back in, file should be there, delete it, log out and back in again.

Arizona5318, 2019-09-08:
On Sept. 8, 2019 Robert M.. wrote...

Hello Andre B...

What does the character (~) proceeding the /Library/Preferences stand for?

I did not include anything prior to the /Library.

Let me know

Cheers

Robert M...

%%%%%%%%%%%%%%%$$$$$$$$$$$$$$$$$$$$$$$$$

https://forum.qt.io/post/203260

kilrah, 2019-09-09:
It means "your home directory".
If you don't put it you get into the "global" system preferences folder. With it you get into your own user account's system preferences folder. The companion preferences are in the latter.

Arizona5318, 2019-09-09:
On Sept. 9, 2019 Robert M..  wrote..

Hello Andre B...

Yes, I did go to the library directory noted in your email and found  the culprit file.

Before that here is what I did to  test out my Apple MacBook Pro system.

Here is another way to test your computer to see if it something more serious than the rouge Companion file you mentioned.

Start a new user account (such as guest Open T), download a copy of Open-Tx.dmg (2.2, 2.2.2, 2.2.3 or 2.2.4) in to the new user account then install it.  Start up Companion that is in the new user account.  If it starts and runs properly then you con go looking for the problems with old Companion version in your "home user account",  in the meantime just use the good copy in the new user account.  If you can not determine the issues with the corrupted Companion version either leave it there until a fix is published or drop it in the Trash Bin and be done with it.  Eventually someone will determine the issues with Companion, at that time you can go back and fix up the original copy of Companion.

That you for your assistance.

Cheers

Robert M..

bsongis, 2019-09-10:
For Raphael ?

raphaelcoeffic, 2019-09-11:
@bsongis looking at it right now...

raphaelcoeffic, 2019-09-11:
Are there any other examples of the sich fields which are not shown on Mac?

raphaelcoeffic, 2019-09-11:
The language menu itself is basically there, but it gets appended to another menu (preferences, at the „leftest“)

raphaelcoeffic, 2019-09-11:
But frankly, spending an hour with lldb, I'm no smarter than I was before....

Here is what I see when it crashes:
```
[D] translations.cpp::Translations::installTranslators():102 - Locale name: "cs_CZ" language: "čeština" country: "Česko"
[D] translations.cpp::Translations::tryLoadTranslator():159 - Installed translation file "qt" from "/usr/local/Cellar/qt/5.12.2/translations" for "cs_CZ"
[D] translations.cpp::Translations::tryLoadTranslator():159 - Installed translation file "qtbase" from ":/translations" for "cs_CZ"
[D] translations.cpp::Translations::tryLoadTranslator():159 - Installed translation file "companion" from ":/translations" for "cs_CZ"
2019-09-11 16:13:36.987 companion[67853:34595267] -[__NSCFType update]: unrecognized selector sent to instance 0x7fdfacccf150
[W] ::unknown():0 - Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

2019-09-11 16:13:37.008 companion[67853:34595267] WARNING: The same NSModalSession was stopped twice. The app may be mixing endSheet:returnCode: with NSModalSession APIs. This is likely to break in the future.
Bus error: 10
```

Not clear which kind of bug this is...

raphaelcoeffic, 2019-09-11:
Seems this is not restricted to Czech or the language menu:
[image]
[image]

kilrah, 2019-09-11:
[image]

I think I got it: 2 menu items have the same exact name, and that gets mixed up?

raphaelcoeffic, 2019-09-11:
Possibly, no clue how that works 😟