opentx/opentx #6724: TH9x with AR9x Board  "hangs" at startup logo.

OldGazer, 2019-09-08:
Radio works fine running Er9x and latest OpenTx 2.2.4.  Flashed with latest 2.3.0N80 nightlie, radio hangs at startup.  I have tested this with an SD card and without an SD card.  I have done a full chip erase and flashed  2.3 using SAM-BA, and the radio does not boot or display the startup logo.

Oddly enough I can put the AR9X into bootloader mode and I can flash with 2.2.4 from an SD card and the radio works.

sreichholf, 2019-11-11:
Same for me, AR9X doesn't boot with 2.3.x here

OldGazer, 2019-11-11:
At this point its obvious to me that this is most likely caused by a code change that the Dev team could not test due to  a lack of hardware.

3djc, 2019-11-11:
> At this point its obvious to me that this is most likely caused by a code change that the Dev team could not test due to a lack of hardware.

Indeed

sreichholf, 2019-11-11:
I'll see if I can find some time (asap) to bisect it

sreichholf, 2019-11-20:
This should be fixed with 0ebf9dbb1e020b24b93d2e3e8c6437999b1bc02f

OldGazer, 2019-11-20:
This is totally AWESOME!!!

What is the ETA for this to it the 2.3.3 nightlies?

OldGazer, 2019-11-21:
I just flashed my Th9x with 2.2.3N49 and IT'S ALIVE!!!

Thanks so much for getting this resolved.  You guys ROCK!!!

closed 2019-11-21 by OldGazer