opentx/opentx #6735: Special Functions reset the Xlite Pro

Talha909, 2019-09-10:
**Describe the bug**
When I try to add special function for the RSSI play out.  My Xlite Pro resets itself. And give error of  No sound on Sd card. After that I switch off the the xlite pro and try to add function again the sound files appears but when i select the file then again xlite pro turn on and off.  And after reset there is some line break shows on the right upper side. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Special Functions
2. Click on Play Track
3. Scroll down to Try to select the voice file
4. See  System reset and next gave error No sound on Sd

**Expected behavior**
Play sound of the rssi
**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.3 or N84)
 - Hardware (e.g. Frsky xlite pro)
 - OpenTX Companion (e.g. 2.3 on windows 10
[image]
[image]
[image]
[image]

)

3djc, 2019-09-10:
What firmware version is your radio running ?

Please post the content of your sound directory as a zip so we can try to reproduce

bsongis, 2019-09-10:
Duplicate of #6730 (scrolling issue)

closed 2019-09-10 by bsongis