opentx/opentx #6737: No R9M Access option on X9D+ SE (Old model) with R9M (old module)

CIXFPV, 2019-09-11:
Hello guys,

Using last OTX 2.3 N84 we can't select the R9M access option on the X9D+ SE (old model).

Bug has been confirmed on 2 radio.

Thanks :)

3djc, 2019-09-11:
When you say old module, I guess you mean 'non 2019' R9 module. Those are not access compatible I believe, only the R9M 2019 are (and lite), so you can't do access with those anyway.

As for radio support, currently only X10 Express and X9D+ 2019 support R9M 2019 in ACCESS mode, we will assess if the other radio ahave the hardware capabilities to support it

bsongis, 2019-09-11:
The X9D+ doesn't have a hardware serial port and the R9M ACCESS need to go to 450kHz
Only X9D+ 2019 and X10 Express are compatible with R9M ACCESS (full size) today

closed 2019-09-11 by bsongis

RRAlex, 2019-09-13:
Just to get this straight as I think a lot of people are confused since FrSky promised things it might not deliver, but will regular ACCESS still be possible with the internal module on "old" X9D+SE?
(without spectrum analyzers or fancy features, but just core control...)

kilrah, 2019-09-13:
Nobody knows but FrSky, and they have not been answering such questions. So assume no.

3djc, 2019-09-13:
That is not currently the case, and we have no information suggesting it will change (or not). Only FrSky would know