opentx/opentx #6738: Vario and Atimeter have stopped working.

CliffEvans, 2019-09-11:
The issue tracker here is for devlopment of OpenTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/) or
[openrcforum](https://www.openrcforums.com/forum/). See also our
[manual](https://opentx.gitbooks.io/manual-for-opentx-2-2/).

For quick questions you can also join our general chat.

After updating to 2.2.4 I am not getting any data from the vario or altimeter on any of my models.  Alt just reads "0" all the time.  
Issues that fall into the category of asking for help, will be
generally closed as invalid without an answer.

3djc, 2019-09-11:
Well, no detail on radio used, no detail on where you where comming from, thats gonna be real hard to investigate. Please post your otx so we can have a look, but that's the first time someone reports something like that in over 40k 2.2.4 downloads

CliffEvans, 2019-09-11:
So sorry!  Horus X10s   otx is in the zip file.  The model I was flying when I noticed this is the SG38  As stated all was fine until I updated!

[Cliff.zip](https://github.com/opentx/opentx/files/3599785/Cliff.zip)

3djc, 2019-09-11:
Any specific reason to have your instance to 25 on that sensor ?

CliffEvans, 2019-09-11:
No, I don't really understand what that means, or how to change it.  I have never changed anything but the firmware.

CliffEvans, 2019-09-11:
I have just fired up the T21 model, Alt just stays on 0 all the time.
[image]

CliffEvans, 2019-09-11:
![20190911_095043](https://user-images.githubusercontent.com/18718856/64682646-b982d000-d479-11e9-91a7-d7129d71770c.jpg

3djc, 2019-09-11:
In your OTX, those sensor have an instance of 17, which is rather unusal, could you try to either rediscover sensors or changing their instance to 1 ?

CliffEvans, 2019-09-11:
Have done all of that. two or three times now!

CliffEvans, 2019-09-11:
Have just changed from 27 to 1, will just go out and try it.

CliffEvans, 2019-09-11:
[image]  instance set to "1"

CliffEvans, 2019-09-11:
[image]
After rediscovering.

3djc, 2019-09-11:
Strange that they are on 27, but there are here and working since you have the *

CliffEvans, 2019-09-11:
I am totally clueless!!!

CliffEvans, 2019-09-11:
I have just downloaded and installed the firmware again, all is now working!!!

closed 2019-09-11 by kilrah

kilrah, 2019-09-11:
You likely had the "faimode" option enabled in the previous one you downloaded.

CliffEvans, 2019-09-11:
I don't know!  As far as I am aware, I have never changed anything but, easily done I suppose.
Thanks to all for the help.