opentx/opentx #6745: {

0crap, 2019-09-11:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

0crap, 2019-09-11:
[image]

notlebowski, 2019-09-11:
That's RSSI-
The lowest value detected....

0crap, 2019-09-11:
I know what RSSI- is.
It gives you the lowest value of RSSI seen in the entire flight.
On the bench next to the TX it should give something like 87 as in this example.
It's now stuck at zero, always.

Yes, it gives sane values in 2.2.1 which I used for a long time.
Also worked fine in 2.3.0N63, from that I jumped on RC1 which gives the issue.

3djc, 2019-09-12:
Technicaly it is not wrong at all, but agreed not very usefull if RSSI- always reports 0

0crap, 2019-09-13:
@bsongis Can you please reopen this issue? I just tested RC2 and the problem is still there.
[image]

3djc, 2019-09-14:
We fixed it and it seems indeed tro have reappeared

0crap, 2019-09-14:
Thx, let me know if you want more sensors tested.

0crap, 2019-09-14:
Tested and fix confirmed.
Thx gents!
Closing this one.

commit ee2dce97 (2019-09-14): Fixes #6745 (not pushed to the right branch!)
commit 1b8e86a2 (2019-09-19): Fixes #6745 (#6757)
commit 6176dd88 (2019-09-19): Fixes #6745 (#6756)
closed 2019-09-14 by 0crap