opentx/opentx #6747: Swedish as a menu language does not work

eral00, 2019-09-11:
Junk characters on display, when I use Menu language Swedish

**Screenshots**
[image]

Testing on ,Horus X10s Express
 - OpenTX version  2.3.0.N85
- Firmware version 2.3.0.N85

3djc, 2019-09-12:
We will fix this, but you need to be aware we haven't had updates  to swedish translation for a looonnng time

3djc, 2019-09-12:
OK, looks like a font issue. Could you check those screen to see if the right characters seem to be used ?

[image]

[image]

PR #6750 (2019-09-12): Fix swedish font
Ideally, this whole  section should be looked at, but this fixes #6747