opentx/opentx #6762: Jumper T12 FlightMode does not change from LogicCondition

PasiRuo, 2019-09-13:
L01=true does not change flightmode if switch=L01

Problem with JumperT12 and otx2.3rc1 is that logic condition does not change flightmode.

3djc, 2019-09-13:
Can you post your otx ? I just tested here, it works fines

PasiRuo, 2019-09-13:
[document12.zip](https://github.com/opentx/opentx/files/3609179/document12.zip)
posted this before to chat.opentx and got confirmation and request to post this report here.

kilrah, 2019-09-13:
Nothing to do with flight modes specifically, just export of switches for T12 has shifts

[image]
[image]

kilrah, 2019-09-13:
Also sources, GV1 comes as CH31

closed 2019-09-13 by PasiRuo

redeyejoe, 2019-10-05:
I was trying to flash opentx to t12 today and file it imports wont open after install. also get error that file doesnt match current version. I had to flash back to the jumpertx firmware. Controller wont boot, just vibrates. It tried to open when I went back into bootloader mode.

PR #6764 (2019-09-13): Fix T12 companion import
This fixes #6762