opentx/opentx #6765: RX #: false positive "same number" detection

LapinFou, 2019-09-13:
Looks like the RX number detection doesn't make any difference of which module is used.

In the example below (red & blue rectangles), Companion doesn't make the difference between the internal XJT module and the external multi module.
[image]

3djc, 2019-09-13:
fixed in rc2

closed 2019-09-13 by 3djc

notlebowski, 2019-09-14:
Please reopen:

I have this in RC3

These models have RxNum #1 

#3  Kinetic  D16
#12 Me109 Multiprotocol
#15 Corsair D8

#15 and #1 are false positives. The Kinetic uses a D16 protocol and the #15 Corsair uses D8
The #12 (Me 109) is indeed not flagged anymore as a duplicate because it's another module

PR #6767 (2019-09-13): Only check rxnum on same module
This fixes #6765
PR #6781 (2019-09-16): check unique RX# based on protocol
fixes #6765 

This should be enough now. We could of course start comparing the subType if multi-proto module is used but well...