opentx/opentx #6770: SDCard content for 2.2.4 (2.2V0021) - 9XR Pro - only 2.2V0018

gweep, 2019-09-13:
Using Turnigy 9XR Pro with OpenTX 2.2.4 using sdcard/opentx-9xrpro/sdcard-9xarm-2.2V0018.  2.2V0021 is not available.

Getting SDCard Error every time I turn on transmitter.    Do I roll back to OpenTX 2.2.3 or ignore error and use 2.2V0018 until 2.2V0021 becomes available?   

I am assuming current config is safe to fly?

SantiCO19, 2019-09-14:
I think the disclaimers state that nightlies should not be used for flying.. With that said, I'm not certain, but I believe I read that you can rename the file found in the sd card root, which identifies the sd card content version, to whatever the applied firmware is expecting to see. This may eliminate the warning you reference. I cannot tell you what additional issues may occur, or be compounded, by this action, if any.

gweep, 2019-09-14:
2.2.4  (2019-07-11)  is a minor release and was downloaded with the "Use releases (stable)" option.   
I will look into your suggestion.

3djc, 2019-09-14:
The file has been added, sorry about that

closed 2019-09-14 by 3djc