opentx/opentx #6771: 2.2.3rc1 and rc2 ACCESS Binding and OTA Update doesn't work anymore

TorbanDallas, 2019-09-13:
Hi, with the nightly 2.2.3N83 r9mlitepro and r9m access (r9m 2019 module) registration and binding was possible, also OTA Update. Now with 2.2.3rc1 and even rc2 on the binding or ota update process the window "waiting for rx" disappears when I power the RX, the radio doesn't respond to any button. Binding and Ota update isn't possible. 
And it has worked pretty good bevor as you can see in my video about it: https://www.youtube.com/watch?v=4_sc-Vc5F1k

kilrah, 2019-09-13:
What receiver

TorbanDallas, 2019-09-13:
here's a short video about the issue
https://youtu.be/SIfRGIPvVag

fape, 2019-09-13:
I think there is missing `menuVerticalPosition = 0;` from stdlcd/radio_sdmanager.cpp REFRESH_FILES() function. It was in before 465674ca689c00d7a9ac7d0f3c5a447e79e64637
see: https://github.com/opentx/opentx/commit/465674ca689c00d7a9ac7d0f3c5a447e79e64637#diff-62acc2d885031a07ee58209d587fd7bf

jodibb, 2019-09-13:
I have same issue. Also doesn't work in latest nightly. (Xlite)

SantiCO19, 2019-09-13:
X-Lite Pro (2.3.0RC1 - opentx-xlites) - R9M Lite Pro/R9MM latest firmware 1.1.2, OpenTX 2.2.3RC1 - For me, registration completes with no apparent issue, but upon binding, the "waiting for rx..." message dissapears, and Rec1 name changes to "---" and the X-Lite Pro begins vibrating in a continuous fashion. The only way to recover is by holding the power button on the X-Lite Pro until is shuts down. I get the same behavior with RXSR ACCESS FPort and Internal Module.

Each attempt to update to RC2 in OpenTX ends with "Compilation Error." Profile settings are correct. I followed the same update process that I have with each previous nightly build, which were all successful.

kilrah, 2019-09-14:
>Profile settings are correct

That doesn't help us, we can't fix it until we know WHAT profile settings cause it.

jodibb, 2019-09-14:
I found temporary workaround: downgrade to opentx from frsky-rc web page, bind, update back to rc02.

0crap, 2019-09-14:
@TorbanDallas Your title contains a wrong OpenTX version number.
It's not 2.2.3 but 2.3.0
You can edit the title if you want.

TorbanDallas, 2019-09-14:
> @TorbanDallas Your title contains a wrong OpenTX version number.
> It's not 2.2.3 but 2.3.0
> You can edit the title if you want.

I did, also on my first comment i made this mistake and even in my video ... but I always mean 2.3.0

closed 2019-09-14 by bsongis

SantiCO19, 2019-09-14:
> 
> 
> > Profile settings are correct
> 
> That doesn't help us, we can't fix it until we know WHAT profile settings cause it. Please post a screenshot.
Lua and noheli were checked in my profile settings. I managed to work around the error by simply unchecking noheli, then RC2 compiled and flashed without further issue. At that point, RC2 showed similar behavior, though repeating slight variations of recovering from the haptic lockup (which additionally now included a sustained "bind-chirp" sound,)  yielded a partial bind. ie RSSI and RxV telemetry displayed, and tel. and RF voice alerts were triggered, but no control data passed to Rx, and still "---" for the Rec1 parameter. Im new to this, so my serialrx settings in BF and my wiring I can't be confident are correct. With this X-Lite Pro, I've never had a reliably working point of reference to make the associations/learn, or confirm that I've correctly carried out the steps provided in documentation.  I will try the workaround mentioned above, and RC3. As well, will aim to include more helpful information in the future.

3djc, 2019-09-14:
There is a fix for that in RC3

SantiCO19, 2019-09-14:
> 
> 
> There is a fix for that in RC3

Thank you. I skipped the workaround, and installed RC3. Bind successful, displaying R9Mini for Rec1. Still no control data going to Rx, but I suppose that is some misconfiguration of mine in BF, or wiring. I mean, there's only 1 wire for Fport! ..but I don't quite have the hang of inverting the inverse of the inverted inversion. Cheers-

TorbanDallas, 2019-09-15:
tested 2.3.0rc3 ota update and binding (with my r9 mini ota) works fine now!

commit 698a7ce1 (2019-09-14): Fixes #6771
commit a1282196 (2019-09-14): Fixes #6771