opentx/opentx #6772: Fry Sky Taranis x9D+

ThePhantom05, 2019-09-13:
Tell me, will firmware 2.3 on Fr sky Taranis x9D Plus be? And then I download copyman 2.3 (OpenTX 2.3.0RC2 (2019-09-13)) and there is no firmware for my Fr Sky Taranis X9D Plus ....

kilrah, 2019-09-13:
This is not a support site but a bug tracker, please put your usage questions on forums.

Yes 2.3 supports the X9D+.

closed 2019-09-13 by kilrah

ThePhantom05, 2019-09-13:
Sorry, I figured out the companion needed to set the firmware - realese channels - realese candidates (testing). 
But now at the moment there is no multi tab, you will have to wait until it appears....

ThePhantom05, 2019-09-13:
Зачем закрыл?

3djc, 2019-09-13:
ABSOLUTELY NOT, multi is now always part of any 2.3 firmware

ThePhantom05, 2019-09-13:
> Это не сайт поддержки, а багтрекер, пожалуйста, задавайте вопросы по использованию на форумах.
> 
> Да, 2.3 поддерживает X9D +.

Зачем закрыл? А?

3djc, 2019-09-13:
English only please

ThePhantom05, 2019-09-13:
Wow, thanks) For the answer)) So you can upgrade)))?

ThePhantom05, 2019-09-13:
The multi has a lot of useful features !!!

ThePhantom05, 2019-09-13:
> Это не сайт поддержки, а багтрекер, пожалуйста, задавайте вопросы по использованию на форумах.
> 
> Да, 2.3 поддерживает X9D +.

Why did you close it ??? BUT

3djc, 2019-09-13:
Can you READ ????? MULTI is IN ALL 2.3 FIRMWARES