opentx/opentx #6773: dockerfile uses gcc-avr version that no longer exists

joba-1, 2019-09-13:
*Before submitting,  search the open issue if there is already a report of the bug*

found no open issues for dockerfile

**Describe the bug**

I try to build latest stable release version 2.2.4 in a supported environment following [OpenTX-2.2-Linux-Build-Instructions](https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions)

This is not possible.

**To Reproduce**

I installed docker on my opensuse leap 15.1 system (any other source system will do)
```
sudo zypper in docker
sudo systemctl start docker
```
Then I downloaded the dockerfile and tried to build the docker image.
```
mkdir opentx
cd opentx
wget https://github.com/opentx/opentx/raw/2.2/radio/util/Dockerfile
docker build -t opentx
```
This fails after some time
```
...
Step 6/10 : RUN wget -q http://ftp.de.debian.org/debian/pool/main/g/gcc-avr/gcc-avr_4.7.2-2_amd64.deb &&     wget -q http://ftp.de.debian.org/debian/pool/main/a/avr-libc/avr-libc_1.8.0-2_all.deb &&     wget -q http://ftp.de.debian.org/debian/pool/main/b/binutils-avr/binutils-avr_2.24+Atmel3.4.4-1_amd64.deb &&     wget -q http://ftp.de.debian.org/debian/pool/main/m/mpclib/libmpc2_0.9-4_amd64.deb &&     dpkg -i gcc-avr_4.7.2-2_amd64.deb libmpc2_0.9-4_amd64.deb avr-libc_1.8.0-2_all.deb binutils-avr_2.24+Atmel3.4.4-1_amd64.deb
 ---> Running in 74a3a025a543
The command '/bin/sh -c wget -q http://ftp.de.debian.org/debian/pool/main/g/gcc-avr/gcc-avr_4.7.2-2_amd64.deb &&     wget -q http://ftp.de.debian.org/debian/pool/main/a/avr-libc/avr-libc_1.8.0-2_all.deb &&     wget -q http://ftp.de.debian.org/debian/pool/main/b/binutils-avr/binutils-avr_2.24+Atmel3.4.4-1_amd64.deb &&     wget -q http://ftp.de.debian.org/debian/pool/main/m/mpclib/libmpc2_0.9-4_amd64.deb &&     dpkg -i gcc-avr_4.7.2-2_amd64.deb libmpc2_0.9-4_amd64.deb avr-libc_1.8.0-2_all.deb binutils-avr_2.24+Atmel3.4.4-1_amd64.deb' returned a non-zero code: 8
```

**Expected behavior**

Image build succeeds.

I already checked: the gcc-avr_4.7.2-2_amd64.deb does not exist (anymore) at http://ftp.de.debian.org/debian/pool/main/g/gcc-avr.
Oldest available version is the 5 year old version 4.8.1.
Since I don't know debian, I don't know where to find older versions.
Since this is my first contact with opentx and the link above warns about inconsistent behavior I hesitate to use other gcc-avr versions.
Do you have a workaround? This should then probably be documented at the link above and dockerfile adapted.

Thank you.

kilrah, 2019-09-14:
You should be able to remove all the AVR lines 16-20 from the Dockerfile since we dropped support for AVR stuff in 2.2 anyway.

joba-1, 2019-09-14:
found the missing files. Hope it helps. It built fine for me now. Thanks @kilrah

closed 2019-09-14 by joba-1

kilrah, 2019-09-15:
Did you try removing them as I suggested? If that works it is a better fix.

joba-1, 2019-09-15:
I didn't go that route first because I didn't want to remove a feature from a stable version, even if that feature is declared not supported.
And I have no means of actually testing the firmware, I wouldn't even know what to look for.

But I did test building it right now: as expected, this works for me too: "[100%] Built target firmware" :)

Thanks!