opentx/opentx #6775: [2.3.0 RC3 - Horus X12S] - RAS always remains at 0

robustini, 2019-09-14:
On my X12S the visualization of the RAS in the statistics is always of value 0, even changing antenna (INT or INT + EXT).
I am in a particular situation, that is I installed the IXJT + RF module of the X10S (triple antenna), so I can't say it's a bug because maybe the firmware doesn't correctly interpret what comes from the module because it expects to find an IXJT and not a IXJT +.
It would still be welcome if he could intercept it anyway.
In this condition with OpenTX V2.2.4 the RAS is displayed only with INT + EXT, it remains at 0 on the inside, but on the outside it reports it.

[image]
[image]

robustini, 2019-09-24:
@raphaelcoeffic @kilrah The problem is not so much that the RAS is not displayed, but that if an antenna has a problem the RF module continues to transmit, compromising its functioning.
I tried now to remove the external antenna for a moment after enabling it and OpenTX didn't notice.

kilrah, 2019-09-24:
>  if an antenna has a problem the RF module continues to transmit, compromising its functioning.

That is always the case anyway.

robustini, 2019-09-24:
> That is always the case anyway.

With OpenTX V2.2.4 if the external antenna is badly fixed the RAS goes to the stars and the radio warns me.
With 2.3 it does not, being always at 0.

kilrah, 2019-09-24:
Sure, but your post suggested it would stop transmitting automatically which has never been the case.

We will likely need some clarification from FrSky about when it can be used/displayed at this point.

Unfortunately we can't just always display it since there are module hardware/software versions and/or antenna selection settings that lead to incorrect/invalid RAS to be sent that we have to ignore, something likely got broken in that logic but we don't know what exactly it's supposed to be.

robustini, 2019-09-25:
> Sure, but your post suggested it would stop transmitting automatically which has never been the case.

I'm sorry Andre, i've explained myself wrongly, I do not mean absolutely that it has to stop transmitting, but simply warn as in 2.2.4, If the user is warned he can do something, if he is not warned he risks compromising the module.

PR #7392 (2020-02-20): Bring back RAS to Horus
Those were having it in 2.2.x, so I guess that should still be ok to have it. 

@bsongis please advise

This closes #6775
closed 2020-02-20 by 3djc