opentx/opentx #6778: [2.3.0 RC3 - X9D+ SE 2019] Crossfire scripts very slow

Target0815, 2019-09-14:
**Describe the bug**
With current Crossfire version (2.93) the LUA scripts for Crossfire are very slow. With Crossfire version 2.41 this was still ok.

Looks to me like the Crossfire scripts still need to be updated.

**To Reproduce**
Update Crossfire to lastest release version (2.93).

3djc, 2019-09-14:
To be sure I understand, is it RC3 releated, or Crossfire version related ?

3djc, 2019-09-14:
OK, tried on x7:
- 2.2.4 with 2.2 scripts and 2.41 : fast
- 2.2.4 with 2.2 scripts and 2.93 : slow
- 2.3.0RC3 with 2.2 scripts and 2.41 : fast
- 2.3.0RC3 with 2.3 scripts and 2.41 : fast
- 2.3.0RC3 with 2.2 scripts and 2.93 : slow
- 2.3.0RC3 with 2.3 scripts and 2.93 : slow

So it is definitively  a change in 2.93. Will contact TBS

Target0815, 2019-09-14:
Thank you. I posted it here because the Crossfire scripts are also "delivered" with OTX.

3djc, 2019-09-16:
Had a discussion with TBS on the subject. On the newer version, a LOT more data are exchanged between TX and RX, specialy when entering this screen.  It's not doing 'anything', on the contrary. They are looking at ways this could be improved, but that not the script or OpenTX, just XFire doing more and more things version after version

closed 2019-09-16 by 3djc