opentx/opentx #6780: Auto-offset for Altitude seems to be not working anymore

notlebowski, 2019-09-14:
On my DLG's I will reset the altitude to 0 whenever ABS(alt) > 1m
Upon binding the system will reset the altitude to 0 and my graphs always looked good.

This weekend I'm seeing this in between flights.
Upon binding it gets a large negative height which spoils my graphs.
It never did this before. Not on <2.2.4 and not on the nightlies.

Today I flew with 2.3.0RC2

[image]

fritzster, 2019-09-14:
I have the same issue since going to 2.3RC. Reset telemetry shows 0 altitude for 1 second and then it just to 291m. Have deleted and re discovered sensors. The altitude is working but with the 291 offset. Have added -291 to the offset for an interim fix.

3djc, 2019-09-17:
What do you use, we don(t seem to be able to replicate,  but we don't have to rebind every flight either.

notlebowski, 2019-09-17:
The word "binding" is used as in "establishing the link".
We don't rebind every flight neither...

Enabling/Disabling logging of my BAMF DLG is a bit complex, so I prefer to take my ELF as an example. I flew it on 7 september and on 14 september with the same setup.
On September 14th I flew it with RC2 and on August 31st I flew it with a 2.3.0 Nightly.

Logging is controlled by this:

L18 Telemetry AND ---- Delay(10s)
L19 !Telemetry AND ---- Delay(10s)
L20 Sticky(L18, L19)

SF1 L20 - SD Logs (0.3s)

You can see the Altitude was reset, but it drifted to -3.4m before it started logging.

Date,Time,VSpd(m/s),Alt(m)
2019-08-31,14:21:47.640,-0.2,-3.4
2019-08-31,14:21:47.850,-0.4,-3.5
2019-08-31,14:21:48.150,-0.2,-3.6
2019-08-31,14:21:48.450,-0.1,-3.6
2019-08-31,14:21:48.750,-0.2,-3.6
2019-08-31,14:21:49.050,0.0,-3.6
2019-08-31,14:21:49.350,0.1,-3.5
2019-08-31,14:21:49.650,0.0,-3.5

This is the exact same setup 2 weeks later

Date,Time,VSpd(m/s),Alt(m)
2019-09-14,14:17:03.150,0.0,-148.6
2019-09-14,14:17:03.160,0.0,-148.6
2019-09-14,14:17:03.540,-0.1,-148.8
2019-09-14,14:17:03.840,-0.1,-148.7
2019-09-14,14:17:04.150,0.0,-148.7
2019-09-14,14:17:04.450,-0.1,-148.8
2019-09-14,14:17:04.750,-0.1,-148.9
2019-09-14,14:17:05.050,0.0,-148.8
2019-09-14,14:17:05.350,0.0,-148.7
2019-09-14,14:17:05.650,0.0,-148.8
2019-09-14,14:17:05.950,0.0,-148.9
2019-09-14,14:17:06.250,-0.2,-149.0
2019-09-14,14:17:06.550,-0.1,-149.0

notlebowski, 2019-09-17:
@fritzster 
Upcoming Friday evening is the first time I can test this again.
Can you check if you can reproduce this with current RC?

fritzster, 2019-09-18:
The offset altitude issue I had was while running ACCESS on my g-rx6 and 2.3RC1 on my 2019 X9D SE it did not happen on the 2.3 nightlies. I’ve since downgraded the RX to ACCST due to sensor loss issues and also upgraded to 2.3RC3 and I no longer have either of the issues.

notlebowski, 2019-09-18:
I had this issue last Saturday with both of my models. It happened during normal operation with 2 different receivers running ACCST that were without issue before. The word "binding" was not used properly by me when I opened the ticket.

I have been daily updating my radio during the holidays and flying a few of my models. Now these holidays are over and I only have access to them during the weekends. Hopefully it was some fluke in RC2. Upcoming Friday evening I will test the radio again. I hope there is no final release before that, but I think there is.

notlebowski, 2019-09-20:
Last Sunday I flashed RC3 on my X-Lite, but never tested it. Just now I tested it with that version and it seems the issue was exclusive to version RC2

3djc, 2019-09-21:
So this is nopt an issue anymore ?

notlebowski, 2019-09-21:
No... 
At the the time there was some work on RSSI-
It may have been some side-effect of a tried solution (or not) ;)

notlebowski, 2019-09-21:
BTW
I now control the logs this way on my DLG's

L18 Telemetry AND Duration(0.1s) Delay(10s)
L19 !Telemetry AND ---- Delay(10s)
L20 Sticky(L18, L19)

SF1 L18 - Reset Alt
SF2 L20 - SD Logs (0.3s)

closed 2019-09-21 by notlebowski