opentx/opentx #6786: Companion RxNum does not differentiate between d8/d16

notlebowski, 2019-09-15:
Please reopen:

I have this in RC3

These models have RxNum #1 

#3  Kinetic  D16
#12 Me109 Multiprotocol
#15 Corsair D8

#15 and #3 are false positives. The Kinetic uses a D16 protocol and the #15 Corsair uses D8
The #12 (Me 109) is indeed not flagged anymore as a duplicate because it's another module
[image]

raphaelcoeffic, 2019-09-15:
No need for a new issue, I have it already covered in #6781

notlebowski, 2019-09-15:
If I add things to a closed ticket it stays unclear if they are addressed.
I have also added several comments to the X12S conversion ticket, but nothing is done with it. 

Radio conversion of my setup is not done perfectly. 
This worries me as the very minor glitch on my setup could be a major one on other's. 

Although I understand it somewhat, I think that tickets are closed too fast without giving the OP a chance to confirm it.

3djc, 2019-09-15:
Closing ticket with commits a is basic git mechanism. If after a fix you are still having an issue, open a new one

notlebowski, 2019-09-15:
But it's it fixed as in that it doesn't occur anymore?
Or you've decided to boot differentiate between protocols?

I would fully understand that in case of the MPM, but I think an exception could be made for the native Frsky protocols.

kilrah, 2019-09-15:
A match will only be reported if same number with same protocol on same module.

notlebowski, 2019-09-15:
OK
Full fix. Even better!

raphaelcoeffic, 2019-09-15:
@3djc with the latest patch, it is with the same protocol on ANY module.

kilrah, 2019-09-15:
Then it's wrong becasue the very fact the modules are different excludes a clash...

raphaelcoeffic, 2019-09-15:
Why? Let’s say I have a model on internal XJT/D16/5 and a model on external XJT/D16/5. Wouldn’t they clash? Or is it prevented by the module internal ID?
If this is the case, we can compare module # as well, no problem, but it works this way. Will just exclude a bit more (said case previously).

raphaelcoeffic, 2019-09-15:
Apart from that, comparing just the module is really wrong, as you could exchange modules. Many people switch between MPM and crossfire for instance.

3djc, 2019-09-15:
No they would not clash, the bind is unique to a module

raphaelcoeffic, 2019-09-15:
Ok, so let’s have module AND protocol and finally be done...

kilrah, 2019-09-15:
> Apart from that, comparing just the module is really wrong, as you could exchange modules. Many people switch between MPM and crossfire for instance.

Precisely, if it's a different module you are allowed to have the same RX number without risk of clashing.
The one exception is someone using 2 multis and defineing the same forced ID for both, but then it's their problem (and they likely did it do "deactivate" the whole purpose of having a module number anyway)

raphaelcoeffic, 2019-09-16:
Protocol and module matching has been implemented and merged, will be in RC4.

closed 2019-09-16 by raphaelcoeffic