opentx/opentx #6789: SD card menu / file handling issue - Horus x10 Express running 2.3 RC2

aussiepage, 2019-09-15:
When navigating and playing sound files on SD card the wrong file is played. After this happens radio also shows corrupted file text that clear next time you navigate to the folder.

**To Reproduce**
Steps to reproduce the behavior:
1. Press and hold SYS to enter radio setup.
2. Press page to navigate to SD card folder
3. Navigate to /sounds/en/system folder.
4. File 0000.wav is highlighted.
5. Press enter and play file to hear "Zero" - good :)
6. Navigate down to 0007. wav and press play to hear "Seven" - good :)
7. Navigate to next sound file 0008.wav (causing screen to scroll up one line).
8. Attempt to play 0008.wav and you hear watt1.wav - bad :(. This was the last valid sound file in the folder. Note watt1.wav is selected as soon as Play/Copy/Rename/Delete menu is shown.
9.  Scroll down one more line to see corruption.
10. Press RTN to return to [..] of System sound folder.
11. Select back to show watt1.wav is in fact last sound file in this folder and corrupt lines are now cleared.
12. Continue scrolling backwards to timovr1.wav. Play this file and you hear timer 1 up.
13. Scroll back one more line to timover3.wav to cause the screen list to scroll.
14 Press play but radio always plays timovr1.wav and corrupted lines can be seen again.

In Summary:-
Sound files play correctly if lines have not scrolled on the display - you can play first 8 wave files and last 8 wave files in the folder correctly. All other wave files cause result in wrong file being played and some corrupted text at last lines of text after last file in folder.

Another way to demonstrate issue is select any file where display has scrolled to bring up play menu and you will see the wrong file has been selected in background so copy,rename and delete actions will also use the wrong file.

**General information**
 - OpenTX version 2.3.0 RC2
 - Horus X10s Express

raphaelcoeffic, 2019-09-15:
This has already been fixed and will be available in RC4.

raphaelcoeffic, 2019-09-15:
For reference, this is fixed by #6777.

closed 2019-09-15 by raphaelcoeffic