opentx/opentx #6790: Horus X10 iXJT upgrade not working

LapinFou, 2019-09-15:
Hi folks,

Following the great tutoriel [Extracting Horus internal RF firmware](https://github.com/raphaelcoeffic/horus_frk) of @raphaelcoeffic I generated the attached .frk firmware for a friend (he is running the ACCST LBT)  → [X10-X10S_LBT_1.3.03.zip](https://github.com/opentx/opentx/files/3613646/X10-X10S_LBT_1.3.03.zip)
Unfortunately he is always getting an error message just after the black screen with **"Device Reset"** information.

I don't know if it is a radio bug on the Horus X10 or something else.

I tried the same procedure on a Horus 12XS and it did work perfectly.

3djc, 2019-09-16:
Reproduced

LapinFou, 2019-09-16:
Good!
Thanks. 👍

commit b847db62 (2019-09-17): Fix x10 internal module flashing
This fixes #6790
closed 2019-09-17 by 3djc