opentx/opentx #6796: [SD Card] Adding a LUA script or built-in menu to switch ACCST mode on EXPRESS radios

LapinFou, 2019-09-16:
Hi folks,

The new FrSky radio with an ISRM RF module have got the ACCESS and the ACCST D16 protocols.
The FrSky RF firmware doesn't make a difference between the ACCST D14 FCC and the LBT mode.
The only way to switch from one norme to another one is through a LUA script provided by FrSky.
I know it is working for the X-Lite Pro/S ans the X9-Lite. I don't know for the new Horus EXPRESS and the X9DP+ 2019.

It would be very handy to have this script included in the official SD card pack. Or, even better, a built-in menu in the OpenTX.

Here is the LUA script provided by FrSky:
```
local toolName = "TNS|Change ISRM mode|TNE"

---- #########################################################################
---- #                                                                       #
---- # Copyright (C) OpenTX                                                  #
-----#                                                                       #
---- # License GPLv2: http://www.gnu.org/licenses/gpl-2.0.html               #
---- #                                                                       #
---- # This program is free software; you can redistribute it and/or modify  #
---- # it under the terms of the GNU General Public License version 2 as     #
---- # published by the Free Software Foundation.                            #
---- #                                                                       #
---- # This program is distributed in the hope that it will be useful        #
---- # but WITHOUT ANY WARRANTY; without even the implied warranty of        #
---- # MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the         #
---- # GNU General Public License for more details.                          #
---- #                                                                       #
---- #########################################################################

local mode = -1

local function redrawPage()
  lcd.clear()
  lcd.drawScreenTitle("ISRM MODE", 0, 0)

  lcd.drawText(5 + 28, 30, "LBT / EU", CENTER)
  if mode == 0 then
    lcd.drawFilledRectangle(5, 14, 56, LCD_H - 25, 0)
  else
    lcd.drawRectangle(5, 14, 56, LCD_H - 25, 0)
  end

  lcd.drawText(67 + 28, 30, "FCC", CENTER)
  if mode == 1 then
    lcd.drawFilledRectangle(67, 14, 56, LCD_H - 25, 0);
  else
    lcd.drawRectangle(67, 14, 56, LCD_H - 25, 0);
  end

  if mode == -1 then
    lcd.drawText(LCD_W/2, LCD_H - 8, "Lecture du mode...", CENTER)
  else
    lcd.drawText(LCD_W/2, LCD_H - 8, "Verif. legislation pays!", CENTER)
  end
end

local function modeRead()
  return accessTelemetryPush(0, 0, 0x17, 0x30, 0x0C40, 0xA0AA5555)
end

local function modeWrite(value)
  if 0 == value then
    command = 0xA0AA5555
  else
    command = 0xA1AA5555
  end
  return accessTelemetryPush(0, 0, 0x17, 0x31, 0x0C40, command)
end

local function telemetryPop()
  physicalId, primId, dataId, value = sportTelemetryPop()
  if primId == 0x32 and dataId >= 0X0C40 and dataId <= 0X0C4F then
    mode = math.floor(value / 0x1000000) - 0xA0
  end
end

local function runPage(event)
  if event == EVT_EXIT_BREAK then
    return 2
  elseif event == EVT_PLUS_FIRST or event == EVT_ROT_RIGHT or event == EVT_PLUS_REPT or event == EVT_RIGHT_FIRST or event == EVT_MINUS_FIRST or event == EVT_ROT_LEFT or event == EVT_MINUS_REPT or event == EVT_LEFT_FIRST then
    local newmode
    if mode == 0 then
      newmode = 1
    else
      newmode = 0
    end
    mode = -1
    modeWrite(newmode)
  else
    if mode == -1 then
      modeRead()
    end
    telemetryPop()
  end
  redrawPage()
  return 0
end

-- Init
local function init()
end

-- Run
local function run(event)
  if event == nil then
    error("Ce n'est pas un script modele !!")
    return 2
  end

  return runPage(event)
end

return { init=init, run=run }
```

3djc, 2019-09-16:
Since this script is completely illegal, this script is not publicly provided by frsky, or us

closed 2019-09-16 by 3djc

LapinFou, 2019-09-16:
??? Looks like Frsky send by email this script to one member of the French Taranis forum.
So, I don't see any legal issue.
Despite the origin of the script, how can we switch the ACCST norm on a ISRM RF module ?

raphaelcoeffic, 2019-09-16:
@LapinFou normally you're not supposed to. In the email your friend received, it was probably written that he must not forward it to anyone, just like in the email I received as I asked for it ;-) I think this is a RF regulations issue.

LapinFou, 2019-09-16:
OK, indeed he didn't tell me. 😄 
I deleted the script.
So, does it mean if you buy a FCC ISRM radio, there is no legal way to switch it in LBT? 😨

raphaelcoeffic, 2019-09-16:
The support will help you if you make them clear that you did it "by mistake" and you want to go legal by reprogramming your module to match your region.

LapinFou, 2019-09-16:
Thanks! 👍 
Go to know, specially for European people buying staff directly in China...

kilrah, 2019-09-16:
Normally the dealers have been giving the script to people who are in an FCC region but got the EU version by mistake (especially since... an update mistakenly switched FCC radios to EU) but not the opposite. If you're in EU, buy an EU version.

lshems, 2019-09-16:
As I understood there is no FCC certification yet for this module, so perhaps the retailers should be adviced not to sell it at all to FCC users?