opentx/opentx #6797: Special function Playsound not working 2.3 RC4 Horus x10s Express

aussiepage, 2019-09-17:
Not certain this is a bug but no sounds are played when

1. Program a Play Sound special function to trigger on a switch.
2. For example select Play Sound Beep1.
3. Activate special function via switch.
4. No sound is played. 

In my tests none of the playsound selections work (any of the available sounds).

Also do not get normal chirp when Tx is in bind mode.

Special function Play Track does however work.

**General information**
 - OpenTX version 2.3.0 RC4 
 - Hardware X10s Express

kilrah, 2019-09-17:
Check the "Beep Volume" setting?

aussiepage, 2019-09-17:
Thanks for suggestion.
Checked "Beep Volume" in Radio Setup screen and it was not zero. Increased to max and still no sound.

3djc, 2019-09-17:
Could you share your otx file ? We are not able to reproduce the issue here

bsongis, 2019-09-17:
I just flashed 2.3 RC4 on mine, it works as expected!

aussiepage, 2019-09-17:
Thanks all for testing and apologies this may not be an issue.
Radio was set to Mode  - Alarm and this seems to mute the Playsound.
Changed mode to NoKey and Playsound works.

Many thanks.

closed 2019-09-17 by aussiepage

kilrah, 2019-09-17:
Yes Alarms disables all beeps other than alarms.