opentx/opentx #6798: Companion telemetry conversion artefacts for X9D+

elecpower, 2019-09-17:
Conversion from a X9D+ 2.2.4 otx for any model with discovered telemetry displays Mod. **External** in the first 32 telemetry rows and blank for the remainder.
Models with no telemetry the Mod field is blank for all rows as expected.

End of configured sensors
[image]
Rows 32 onwards
[image]

For a new model created in 2.3.0RC4 all fields are blank prior to any discovery on the radio. I haven't had a chance to flash the radio with 2.3 to test what is displayed after discovery.

**General information**
 - OpenTX version 2.3.0RC4
 - Hardware Frsky X9D
 - OpenTX Companion 2.3.0RC4 on Ubuntu 18.04)

bsongis, 2019-09-17:
Ah yes I saw it in the pull request and forgot to mention it. If the 3 bits are 1, the origin is "SPORT LINE", not the module given in the first bit. We have only 3 bits for storing the module index + rxId ...

raphaelcoeffic, 2019-09-17:
and how is "external module / third receiver" encoded? the same?

bsongis, 2019-09-17:
External module / First receiver = 100
External module / Second receiver = 101
External module / Third receiver = 110

elecpower, 2019-09-17:
The receivers are bound to the radio's internal module. I assumed the module was the  telemetry source which is external to the radio.

[image]

raphaelcoeffic, 2019-09-17:
@bsongis ok, so Companion should show something like "SPORT" in this case? (instead of a module or receiver?) Or maybe even nothing? (hiding the corresponding fields?)

bsongis, 2019-09-18:
Yes, SPORT (or even nothing) will be perfect

raphaelcoeffic, 2019-09-22:
@bsongis seems we another problem here.

In the radio conversion, we set the upper 3 bits to 1 in case of XJT PXX1:
```
    if (oldModel.telemetrySensors[i].type == 0 && (oldModel.moduleData[0].type ==  MODULE_TYPE_XJT_PXX1 || oldModel.moduleData[1].type == MODULE_TYPE_XJT_PXX1))
      newModel.telemetrySensors[i].instance = 0xE0 + (oldModel.telemetrySensors[i].instance & 0x1F) - 1;
```

However, in Companion, we don't set these bit, and mask the instance for all PXX1 modules:
```
        if (model.moduleData[0].isPxx1Module() || model.moduleData[1].isPxx1Module())
          sensor.instance = (_instance & 0x1F) + (version <= 218 ? -1 : 0); // 5 bits instance
```

That does not seem to be consistent.

Shouldn't we set these 3 bits for anything but PXX2 modules?

raphaelcoeffic, 2019-09-22:
Further comment: Why do we subtract 1 for sensors with `instance == 0`? I think this is not correct.

bsongis, 2019-09-23:
1) In EEPROM v218 we don't have ACCESS, so writing 0xE0 when the module is PXX1 seems correct. I think the radio conversion is fine.

2) In Companion EEPROM export / import, we split the instance into (instance / rxIdx / moduleIdx)

3) Instance == 0 is impossible for FrSky sensors in 2.2

raphaelcoeffic, 2019-09-23:
> In EEPROM v218 we don't have ACCESS, so writing 0xE0 when the module is PXX1 seems correct. I think the radio conversion is fine.

The question was more: is `MODULE_TYPE_XJT_PXX1` in the radio equivalent to `isPxx1Module()` in companion? I don't think so, is it?

> Instance == 0 is impossible for FrSky sensors in 2.2

What about the disabled/blank sensors? the issue is that the instance is then `!= 0`, and the Companion thinks the sensor is not disabled. So I think we should not "convert" sensors for which the instance `== 0`, right?

bsongis, 2019-09-23:
a disabled sensor should have no name (not instance = 0)

bsongis, 2019-09-23:
you are right about the MODULE_TYPE_XJT_PXX1 comment

bsongis, 2019-09-23:
Fixed in RC5

closed 2019-09-23 by bsongis