opentx/opentx #6799: Comp 2.3.0RC4   XJT (D8) protocol, RX # still conflicting, is # necessary for D8?

Waldo-P-Barnstormer, 2019-09-18:
Would it be possible to leave the RX # blank in the model list for receivers using the XJT D8 protocol.
Having RX #'s shown is meaningless and gives a user a false sense that there wouldn't be a conflict if numbers are different. That's not the case. Having two D8 receivers powered at the same time they will both respond to the Tx, RX # same or different.
Also in 2.3.0RC4 my model list still shows many conflicts with D8 likely caused by adding models through copying, pasting and editing them over many years.
There is a work around to clear all D8 conflicts. I momentarily change them to D16, assign Receiver No.0 to them and then switch back to D8 and the RX # is cleared in the column. Why that works??, but it does.

To create a conflict make a model using the D8 protocol. Companion assigns a Rx # same as the model slot. 
Create a D16 receiver on another line using previous Rx #, no conflict, but now change it to a D8 type and a conflict shows up. 
They both now have the same Rx # but no way to change them in the D8 setup screen. 
But like I said above models using D8 conflict with the same or different RX #'s.

notlebowski, 2019-09-18:
It could show D8 instead of the RxNum in Companion.
Like Waldo said, these should never conflict.

kilrah, 2019-09-18:
Yes D8 doesn't have RX no so should be ignored

PR #6804 (2019-09-18): RXID should be ignored in D8, like on the radio
Thix closes #6799
closed 2019-09-19 by 3djc