opentx/opentx #6803: Taranis Q X7 double entries with mixers

SjDost, 2019-09-18:
I noticed when reading models from the Taranis Q X7 (rel X7-2.2.4) that double entries are created in the Mixers. It looks that this happens only with Mixers above Ch20.
Companion used is OpenTX 2.2.4 Jul 11, 2019.
Attached are the OTX file and the corresponding ODT and PDF files. The issue is above CH20.
Regards SjDost

[Double mixer lines.zip](https://github.com/opentx/opentx/files/3625785/Double.mixer.lines.zip)

3djc, 2019-09-18:
Could you add you eeprom file ?

SjDost, 2019-09-18:
Sorry but what do you need exactly? The Eeprom file on the SD card is empty
apart from the readme file.
Do you need de firmware files I can download in Companion?
Regards SjDost

3djc, 2019-09-18:
You need to go to that screen and LONG ENT to create an EEprom backup that you will then find in the EEPROM folder of your SD

[image]

SjDost, 2019-09-18:
Added the requested Eeprom Bin file. 
[EEPROM.zip](https://github.com/opentx/opentx/files/3625995/EEPROM.zip)

SjDost, 2019-09-18:
Now the correct EEPROM file requested. Sorry or the confusion on my part..
[eeprom-2019-09-18-125438.zip](https://github.com/opentx/opentx/files/3626028/eeprom-2019-09-18-125438.zip)

closed 2020-06-14 by stale[bot]